nasa/osal #471: Order of operations on OS_DeleteAllObjects

jphickey, 2020-05-19:
**Is your feature request related to a problem? Please describe.**
OS_DeleteAllObjects() is used when shutting down the system, such as after an exception, a commanded processor restart, or CTRL+C, etc.

This simply deletes resources based on their numeric `OS_OBJECT_TYPE` value, meaning tasks (1) are first, followed by queues, bin/count, semaphores, mutexes, etc. and eventually timers (9).

A recent issue described in nasa/cfe#701 observed a potential problem with this.  As the task and semaphore are deleted, a timer could still be running.  If that timer executes during shutdown, it may interrupt the deletion, and attempt to use semaphore objects.

Normally this shouldn't be an issue because OSAL will return an error and reject the call.  However due to an underlying issue in Binary Semaphores (#470) after task cancellation, this caused deadlock.

**Describe the solution you'd like**
It would be preferable to delete timers first, then tasks, then semaphores, files, and other resources.  This should be a safer ordering in general, as it will reduce the potential for resources to be used as they are being deleted.

**Describe alternatives you've considered**
Leave as-is.  

**Additional context**
The real fix for #470 prevents deadlock, this is just more future-proofing.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

CDKnightNASA, 2020-05-19:
It wouldn't hurt to also delete objects in reverse order...For short-lived instances they will likely be in reverse order (eldest first). Of course this is no guarantee, but easy to implement in the code.

skliper, 2020-12-09:
@jphickey want to resolve this for Caelum?

jphickey, 2020-12-09:
Sure, I think I already have a patch for this somewhere that never got put into a final PR.  I can find it and submit.

PR #711 (2021-01-12): Fix #471, order of operations for delete all
**Describe the contribution**
When cleaning up for shutdown, delete resources that have a task/thread first, followed by other resource types.  This helps avoid possible dependencies as running threads might be using the other resources.

Fixes #471

**Testing performed**
Build and run unit tests
Sanity check CFE, ensure shutdown works as expected

**Expected behavior changes**
No visible impact ... internally the tasks are deleted first during shutdown, which only has an impact if/when tasks are actively using other OSAL resources.

**System(s) tes [... cut]
commit 35795cac (2021-01-12): Merge pull request #711 from jphickey/fix-471-deleteall-order

Fix #471, order of operations for delete all
commit b5863b71 (2021-01-12): Fix #471, order of operations for delete all

When cleaning up for shutdown, delete resources that have
a task/thread first, followed by other resource types.

This helps avoid possible dependencies as running threads
might be using the other resources.
PR #750 (2021-01-13): osal Integration candidate: 2021-01-12
**Describe the contribution**

Fix #702, use iterators instead of for loops
Fix #645, implement separate lock/unlock key
Fix #703, implement exclusive/reserved lock
Fix #642, make OS_TaskDelete sync
Fix #703, unit test 
Fix #697, use POSIX dir implementation on V
Fix #580, improve FS_BASED mounts on 
Fix #708, chmod error h
Fix #471, order of operations for delete all
Fix #445, add pointer parameter checks
Fix #573, add OS_FileSysSta
Fix #544, add pointer check
Fix #606, Resolve cast-align error in VxWorks OS_TaskGet
Fix #429, update OSAL code to use time [... cut]
closed 2021-01-13 by astrogeco