nasa/osal #499: Linux BSP shouldn't trim first argument

jphickey, 2020-06-12:
**Describe the bug**
As part of command line processing the logic for Linux BSP is to prune the first argument which is the command name, to be consistent with other (e.g. RTEMS) platforms that do not pass the command name to begin with.

However the CFE PSP passes the argument list to the C library `getopt_long()` and this expects the first argument to be the command name, so this breaks.

**To Reproduce**
Pass arguments (e.g. `-I' to set spacecraft ID value, etc) to start the CFE core on linux.  Observe that the default value is still used, argument isn't processed.

**Expected behavior**
Command line arguments should be processed and used.

Just store/pass the full set of arguments, no real need to abstract, as anything doing argument processing is almost certainly platform-specific (e.g. CFE PSP) so it will know what to expect here.

**System observed on:**
Ubuntu 20.04

**Additional context**
First bug found by newly-minted PSP functional test.  I attempted to use `-I` to force a test failure on the spacecraft ID value and it didn't work.

**Reporter Info**
Joseph Hickey, Vantage Systems, Inc.

skliper, 2020-06-12:
see also nasa/cFS#86

jphickey, 2020-06-12:
> see also [nasa/cFS#86](https://github.com/nasa/cFS/issues/86)

Thanks, I missed this previous bug report on the issue.  Otherwise I would have fixed it earlier...

PR #500 (2020-06-23): Fix #499, store all arguments in generic-linux BSP
**Describe the contribution**

Keep the entire argc/argv from the shell.  Do not prune the command name as getopt expects this to be there.

Fixes #499 

**Testing performed**
Execute unit tests, confirm all passing.
Execute new PSP tests, confirm command line options are being recognized.

**Expected behavior changes**
Command line options in Linux are recognized, no longer ignored/dropped.

**System(s) tested on**
Ubuntu 20.04

**Contributor Info - All information REQUIRED for consideration of pull request**
Joseph Hickey, Vantage Systems, I [... cut]
commit 4878b90f (2020-06-23): Merge pull request #500 from jphickey/fix-499-linux-args

Fix #499, store all arguments in generic-linux BSP
commit f0ebd4a1 (2020-06-23): Fix #499, store all arguments

Keep the entire argc/argv from the shell.  Do not prune the command
name as getopt expects this to be there.
PR #521 (2020-06-25): Integration candidate: 2020-06-17
**Describe the contribution**
Fix #499
Fix #501
Fix #505 
Fix #506
Fix #520
Fix #351

**Testing performed**
Bundle CI - https://github.com/nasa/cFS/pull/106/checks

**Expected behavior changes**
PR #500 - Command line options in Linux are recognized, no longer ignored/dropped.

PR #504 - No impact to current unit testing which runs UT assert as a standalone app. Add a position independent code (PIC) variant of the ut_assert library, which can be dynamically loaded into other applications rather than running as a standalone OSAL application. This enables loadin [... cut]
closed 2020-06-25 by astrogeco