nasa/osal #512: OSAL unit tests terminate before the console thread prints the results

mbenson1, 2020-06-17:
**Describe the bug**
Most of the OSAL unit tests terminate before the console thread is allowed to execute and write the results to the console.

**To Reproduce**
Steps to reproduce the behavior:
1. Build with unit tests, i.e. the initial build created using the instructions in the cFS repository.
2. Run the unit tests manually, not with ctest.  ctest hides the console output anyway.

**Expected behavior**
Running osal_core_UT should look like:

`[BEGIN] PC-LINUX UNIT TEST
OS_printf() - #1 Nominal [ This is the expected stdout output after API call]
OS_printf_enable() - #1 Nominal [This is the expected stdout output after API call]
OS_printf_disable() - #1 Nominal [This is the expected stdout output before API call]
OS_printf_disable() - #1 Nominal [This is the expected stdout output after test reset]

ut_oscore HAS 167 tests total.
ut_oscore PASSED 135 tests.
ut_oscore FAILED 0 tests.
ut_oscore HAS 5 tests requiring manual inspection.
ut_oscore SKIPPED 26 tests relating to OS-call-failure.
ut_oscore SKIPPED 1 tests not applicable to the platform.
`
Instead, depending on the timing of the target under test, you might see:

`[BEGIN] PC-LINUX UNIT TEST
OS_printf() - #1 Nominal [ This is the expected stdout output after API call]
OS_printf_enable() - #1 Nominal [This is the expected stdout output after API call]
OS_printf_disable() - #1 Nominal [This is the expected stdout output before API call]
OS_printf_disable() - #1 Nominal [This is the expected stdout output after test reset]
`

**Code snips**

The easiest fix is to add:

`
    /* Yield to allow the console thread to complete writing to the console. */
    OS_TaskDelay(2000);`

to the end of the function "void UT_os_teardown(const char* testSuiteName)" in file "ut_os_stubs.c".  This will ensure the caller thread pends for long enough time (hopefully) for the console thread to release and complete printing to the console.  However, this is not 100% bullet proof since it still relies on the console thread starting and completing the print in the time provided to the OS_TaskDelay() function.

**System observed on:**
 - VirtualBox virtual machine
 - Ubuntu 16.04
 - OSAL 5.0.3-bv

**Additional context**
Most platforms (all platforms I know of) use a separate, kernel created thread, to print to the console already.  The OSAL provided console thread just adds a third thread to 2 already existing threads.  I'm not sure why the additional thread needed to be added.  But if it wasn't there, this problem is resolved by calling fflush, which pends until the output has successfully been sent out.  Aside from making the OSAL more complicated, I'm not sure what problem the new console thread is solving.

**Reporter Info**
Mathew Benson
Owner
Windhover Labs, LLC

jphickey, 2020-06-29:
I looked into this, I was able to confirm a similar effect locally using a single-core VM.

Probably prudent to inject a small shutdown delay prior to returning from "main" to give a chance for any shutdown-related OS_printf() calls to complete.  This will also cover the FSW use case, not just unit test.   Also worthwhile to do this for the "MIR" test cases which specifically call OS_printf(), to help increase the chance that the output gets into the log file in the expected spot.

For background info, the separate thread isn't really necessary on Linux but it is implemented for consistency with RTEMS and VxWorks.  There is a macro called `OS_CONSOLE_ASYNC` that could turn this off and make `OS_printf()` calls go direct.  It still exists but is currently forced to "true" - not part of user specified config - to reduce the testing matrix.  You can still change it in the code, though, here:

https://github.com/nasa/osal/blob/80fdd67a6c28a962164697ece12570142331ea44/src/os/posix/src/os-impl-console.c#L44

However even without changing this line you still have the option of setting `OS_CONFIG_UTILITYTASK_PRIORITY` to a low numeric value which is basically the same effect, hence why it was seen as not worth the effort to also support an `OS_CONSOLE_ASYNC` option in addition to this.

skliper, 2020-12-09:
OSAL unit-tests have been converted to use the ut_assert library for testing/reporting, which utilizes OS_BSP_ConsoleOutput_Impl, writing directly to STDOUT_FILENO unbuffered.  Should no longer see missing results from the unit test.  Closing as duplicate.

closed 2020-12-09 by skliper