nasa/osal #514: osal-wb-ostimer-test has an incorrect failure

mbenson1, 2020-06-17:
**Describe the bug**
The UT_os_timerset_test() in the osal-wb-ostimer-test calls "OS_TimerSet(99999, startTime, intervalTime)", with both startTime and intervalTime set to 0, expecting that it will fail the ID check before it will fail the start and interval time checks.  

**To Reproduce**
Steps to reproduce the behavior:
1. Build and run osal-wb-ostimer-test on the sample build from the getting started instructions in the cFS repository.

**Expected behavior**
The unit test is written expecting that the ID validity check will fail before the start and interval time check fails.  Its expecting it will return OS_ERR_INVALID_ID, but when when you run this on using the POSIX OSAL from 5.0.3-bv version, it will actually return OS_ERROR because the start and interval times are invalid.  

**Code snips**
A quick fix is to initialize startTime and intervalTime to 1, which will cause the OS_TimerSet() function to pass the start and interval time checks, and fail on the ID check as expected.

**System observed on:**
 - VirtualBox
 - Ubuntu 16.04
 - OSAL 5.0.3-bv

**Additional context**
While changing the start and interval times to 1 will fix this problem in this specific case, the test is actually specific to the OSAL (posix, vxworks, rtems, etc...) under test.  I would argue that a better design would be to have 1 common OSAL unit test to verify functionality, i.e. that tasks function as tasks, and mutexes function as mutexes, but that each OSAL might have their own individual set of unit tests that are specific to that OSAL.  You can think of the common set as black box tests, that are verifying the OSAL meets functional requirements, and white box tests for coverage.  White box testing, by their nature, are specific to the source code under test.  Therefore, each OSAL might have its own set of white box testing to ensure coverage for that particular OSAL.  

We follow this concept in our CFS fork.  The unit tests originally created by GSFC work well as black box tests.  They have little to know platform or configuration dependency.  The unit tests originally created by JSC work well as white box tests, as this bug report indicates.

**Reporter Info**
Mathew Benson
Owner
Windhover Labs, LLC

jphickey, 2020-06-18:
To confirm - I looked up this function in the current "master" branch - is this the code you are referring to:

https://github.com/nasa/osal/blob/bc73437ea95af54078ba40ce4001a567b1bbc393/src/unit-tests/ostimer-test/ut_ostimer_timerio_test.c#L505-L510

This condition, however, is only checking for `OS_ERR_NOT_IMPLEMENTED` - which should not be possible with any of the current OSAL libs - it is implemented in all of them.

Whereas the next call to OS_TimerSet() has nonzero values for the times, and this is where it's checking for `OS_ERR_INVALID_ID`:

https://github.com/nasa/osal/blob/bc73437ea95af54078ba40ce4001a567b1bbc393/src/unit-tests/ostimer-test/ut_ostimer_timerio_test.c#L515-L519

Note we do run all unit test (both white box and black box) as part of our CI cycle and all validation must pass for changes to be accepted to master branch.  The places where things most often fall through is on platforms other than Linux/POSIX where we do not have a CI script to run them on a regular basis.

jphickey, 2020-06-18:
OK - looking back in time at 5.0.3-bv tag, I can see this is the way it was back then:

https://github.com/nasa/osal/blob/155e9ebcd6d1930890231a44237e6883d229d22c/src/unit-tests/ostimer-test/ut_ostimer_timerio_test.c#L604-L608

So in short, looks like this is probably already addressed in the "master" branch.

mbenson1, 2020-06-18:
I corrected my ticket to include the correct name of the unit test.

jphickey, 2020-06-18:
One more note - I consider both the JSC test (`unit-tests` dir) and GSFC test (`tests` dir) to be black box.  Both should work with alternative OSAL implementations.  There is a separate `unit-test-coverage` dir that has white box tests in it, which are tightly coupled with implementation.

The issue arises in some black box tests where the parameters being passed in have _multiple_ incorrect items, such as a bad ID _and_ a bad interval.  The API doesn't specifically say which takes precedence.  Any test case that does this could be considered a bug, because it should only pass one bad value at a time - otherwise the expected return code is not well defined.

mbenson1, 2020-06-18:
Ok, but they don't work with all alternative OSAL implementations.  A change had to be made to the unit test for it to work with the newly updated OSAL from NASA.  That change now causes it to fail on other OSALs that follow the same argument check order that the previous one did.  These may work just fine on the NASA released OSALs, but not other non-NASA released OSALs even though they are compliant with the API.  That's why I say these are implementation specific.

skliper, 2020-06-18:
> Any test case that does this could be considered a bug, because it should only pass one bad value at a time - otherwise the expected return code is not well defined.

Agreed.  These cases should get documented and fixed.

jphickey, 2020-06-18:
Not sure I'm following the current concern -- The original description of passing 0 for the timer values is no longer an issue as the current code is no longer doing this.   I'm not saying this wasn't a bug - it certainly _was_ a bug to pass zero for these time intervals  I'm just saying it's already addressed - unless I'm missing something?

The `5.0.3-bv` is a tagged version and there is no guarantee it is bug free or is there a plan to release patches to this.  Any bug fixes that we do implement will be found in `master` until the next official release, and the fix is already there.

As best I can tell, there is still one valid concern here regarding the use of magic number `99999` as the "bad ID" - the API doesn't really specify what constitutes a good vs. bad ID so in an alternative non-NASA OSAL implementation I can see this being a possible problem.  We probably should introduce a symbol/macro which is guaranteed to always be an "invalid" ID value.

mbenson1, 2020-06-18:
Then the API needs to be better documented.  If that's black box testing, what requirement is that testing?  There is nothing in documentation that states what the legal ranges are for start_time and interval_time.  I checked the code that that unit test was originally written against.  There was no check on the start_time or interval_time.  If that is truly black box testing, there should be a requirement or at least some documentation stating API compliance requires validity checking those arguments.

jphickey, 2020-06-18:
Yes - I agree the documentation is wrong.  It does say you can pass zero as the interval time for a "one-shot" behavior.  This is not used by CFE and actually totally untested as we don't have any test case for it.   This should be corrected.

jphickey, 2020-06-18:
FWIW - my recommendation would be to remove any support for "one shot" timer behavior - CFE doesn't need it and unless there is a specific requirement for it, we should only support interval/periodic timers.

jphickey, 2020-06-18:
Tagging @skiper, @jwilmot , @acudmore  -- can we remove suggestions of "one shot" timer behavior in the API documentation and say that all timers need to have a nonzero time interval for both start and interval time?

skliper, 2020-06-18:
I was just typing up a ping... lets add @ejtimmon @astrogeco as they both interact with some of our major stakeholders.

mbenson1, 2020-06-18:
Slow down.  Please.  Its one thing to change internal code.  Its another to change API requirements.  You are not the only people that these API changes affect.  So before you go making changes to API documentation so that it fits your unit test, please follow a vetted and documented process.  Maybe poll the community and give them a chance to provide feedback to things that might affect them.

mbenson1, 2020-06-18:
Before you make that change and prohibit a zero value for the interval, understand that when you do that, SCH will break.  It relies on that "one shot" capability in 2 places within sch_custom.c code.  The application unit test code was deleted in 6.7, so I assume it wasn't unit tested against it.  Has anybody used SCH with 6.7?

jphickey, 2020-06-18:
Nobody is making the change yet.  I'm just saying what's there right now, and that (AFAIK) there is no existing CFE requirement for one shot timers.  I could be wrong.

Correct - The next step is to poll the community and decide if we actually do need this (and also write a test for it) or just fix the documentation to reflect what is actually implemented.

Also AFAIK we don't have a separate OSAL requirements document, it is mainly driven from the CFE requirements.

jphickey, 2020-06-18:
> Before you make that change and prohibit a zero value for the interval, understand that when you do that, SCH will break. It relies on that "one shot" capability in 2 places within sch_custom.c [...]

Ahh Yes, SCH - particularly the `sch_custom.c` file has a number of issues bigger than this one - that in itself is worthy of its own discussion.  But FWIW that code isn't really part of the core app - merely an example of how it was done once (and a pretty complex example if you ask me).

But you are correct in that SCH is probably the biggest user of OS timers and so needs to be considered with any change here.

mbenson1, 2020-06-18:
This goes along with the email I just sent on the CFS mailing list.  GSFC did a fantastic job designing the PSP/OSAL/CFE/App API, allowing CFS to take on a life beyond just GSFC, Ames, and JSC.  Part of that is because of the well thought out API that has remained stable for at least the 6 years I've been working with it.  There may be other applications developed by other teams that are working to an established API.  I appreciate that you're considering SCH in your decision, but there may be others using that API.  Think of them before you remove capability that you don't need.  Just because you don't think you need that one-shot capability shouldn't be grounds to just remove it.

jphickey, 2020-06-18:
FYI - I did submit a separate issue #515 as I think its worthwhile to fix that regardless of however this requirement issue is addressed.

I can't speak for GSFC, they dictate the requirements for this product (not me - I just voice my opinion which means nothing).  I know its always a constant struggle to balance the needs/expectations of external users and other NASA users.  Lately there has been more of a push to streamline and simplify the code.

skliper, 2020-06-18:
> Before you make that change and prohibit a zero value for the interval, understand that when you do that, SCH will break. It relies on that "one shot" capability in 2 places within sch_custom.c [...]

Yes, but SCH is due an update (it's actually up next in the queue, but CF has been the higher priority).  Many of the NASA apps are being worked right now to modernize relative to the current framework code base. They typically follow the framework updates, and their open source release process is even more difficult to navigate so the open source code typically falls behind the framework by a version or two.

mbenson1, 2020-06-18:
Is there anybody on the CCB that represents anybody outside of a NASA center?  Who is the CCB chair?

skliper, 2020-06-18:
@astrogeco is the current chair.  No outside NASA CCB reps, but all the framework activity now happens on GitHub so the entire community is welcome to comment/review/submit/express concern/etc relative to PRs.  See the cFS wiki for minutes.

skliper, 2020-06-18:
Previous cFS management has explored an option for the extended community to "buy in" to the CCB (either with resources or contributions), but a working model was never finalized.  Typically it's the paying stakeholders that get an official invite and NASA community represented.  Open source is really just best effort without dedicated resources...

skliper, 2020-09-01:
Original issue is OBE.  Spawned #586 to add one-shot timer unit test.  If there's any other actions I missed out of this thread feel free to create an issue, but as far as I can tell that's the only forward action.  API documentation covers the expected behavior and code look good to me.

closed 2020-09-01 by skliper