nasa/osal #515: Use of hardcoded numbers for bad IDs in OSAL unit tests

jphickey, 2020-06-18:
**Is your feature request related to a problem? Please describe.**
Some unit tests pass a hardcoded magic number i.e. 99999 to OSAL functions to evoke a bad ID error, such as the case here:

https://github.com/nasa/osal/blob/bc73437ea95af54078ba40ce4001a567b1bbc393/src/unit-tests/ostimer-test/ut_ostimer_timerio_test.c#L515-L519

As the API doesn't specifically dictate what constitutes a good ID vs. a bad ID, there is a possibility that in some OSAL implementation this value `99999` might actually be a valid ID.  

This is a concern for black box tests which are supposed to be implementation-agnostic, but need to validate that the correct error is returned when called with a bad ID.

**Describe the solution you'd like**
The OSAL API should define macro to provide a value which is guaranteed to always be a valid ID.  Unit tests can use this symbol instead of the magic number.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2020-06-18:
Thinking about this further, any ID which wasn't directly output from a previous call to  a create function (e.g. `OS_TimerCreate()`) should rightfully return `OS_ERR_INVALID_ID`.  That is, it isn't just a simple range check, it is testing that it is actually a valid object that exists in the system.

Which would mean that despite being a hardcoded number the code is fine as is and doesn't need to change, because if the code hasn't (yet) called OS_TimerCreate successfully, there are no valid IDs, and _any_ ID passed here should return `OS_ERR_INVALID_ID`.

skliper, 2020-06-18:
Good point.  Maybe we should test every possible integer just to make sure though (inside joke).

skliper, 2020-12-09:
OBE.  No use of 99999 for an ID, typically uses OS_OBJECT_ID_UNDEFINED to get OS_ERR_INVALID_ID.  Fixed in #568.

closed 2020-12-09 by skliper