nasa/osal #516: Second parameter of pthread_mutex_timedlock should be absolute time not timeout

jonathan-brandenburg-metecs, 2020-06-18:
In src/os/posix/src/os-impl-binsem.c, there are three calls to pthread_mutex_timedlock, each time with &OS_POSIX_BINSEM_MAX_WAIT as the second parameter where OS_POSIX_BINSEM_MAX_WAIT has a value of 2 seconds. because the second parameter of pthread_mutex_timedlock should be an absolute time and is not a timeout, if the mutex is not immediately available, pthread_mutex_timedlock will return with ETIMEDOUT resulting in the OS_BinSemFlush_Impl, OS_GenericBinSemTake_Impl, or OS_BinSemGive_Impl call returning OS_SEM_FAILURE.

Here's a code snippet of the "as is":
    if ( pthread_mutex_timedlock(&sem->id, &OS_POSIX_BINSEM_MAX_WAIT) != 0 )
    {
       return(OS_SEM_FAILURE);
    }

It should look something like:

   struct timespec timeoutTime;
   clock_gettime(CLOCK_REALTIME, &timeoutTime);
   timeoutTime.tv_sec += OS_POSIX_BINSEM_MAX_WAIT.tv_sec;
   timeoutTime.tv_nsec += OS_POSIX_BINSEM_MAX_WAIT.tv_nsec;
   if ( pthread_mutex_timedlock(&sem->id, &timeoutTime) != 0 )
   {
      return(OS_SEM_FAILURE);
   }

Reproducing this issue is tricky as the cFS application has to run long enough to have a mutex not immediately available for the lock. The behavior I'm seeing in my installation is the background task terminates and Ctrl-C can no longer be used to terminate the application. I'm running in a Centos 7 Virtual Box machine.

Jonathan C. Brandenburg
METECS
jonathan.c.brandenburg@nasa.gov

jphickey, 2020-06-19:
Thanks for catching this!  I somehow missed that the timeout here was absolute and not relative.  Kind of annoying that POSIX isn't consistent here.

jonathan-brandenburg-metecs, 2020-06-19:
Glad to help! It is a bit annoying. I read the documentation for that method 100 times and kept seeing "absolute" but it just didn't compute because that's not at all common. Thank you!

PR #518 (2020-06-23): Fix #516, Use absolute timeout for timedlock calls
**Describe the contribution**
The pthread API defines the timeout parameter for the pthread_mutex_timedlock call as an absolute value based on CLOCK_REALTIME.

This introduces a wrapper function to calculate the absolute timeout for this.

Fixes #516 

**Testing performed**
Build and execute CFE, sanity check CFE operation, and run all unit tests.
Coverage is not affected because there is no coverage test for POSIX.

**Expected behavior changes**
The timed lock in binary semaphores should now correctly wait for up to 2 seconds for the lock to  [... cut]
commit 9cfdbffb (2020-06-23): Merge pull request #518 from jphickey/fix-516-absolute-timedlock

Fix #516, Use absolute timeout for timedlock calls
commit a871decb (2020-06-23): Fix #516, Use absolute timeout for timedlock calls

The pthread API defines the timeout parameter for
the pthread_mutex_timedlock call as an absolute value
based on CLOCK_REALTIME.

This introduces a wrapper function to calculate the absolute
timeout for this.
closed 2020-06-25 by astrogeco