nasa/osal #532: Set posix thread names using task name when creating CFE tasks

dsburns, 2020-07-06:
**Is your feature request related to a problem? Please describe.**
It would be helpful for the pthread name to match the CFE task name in the underlying OS on POSIX platforms.  This would allow for a better user experience when debugging CFE applications using Linux based utilities i.e. strace, htop, etc.

**Describe the solution you'd like**
This can be accomplished by using the _pthread_setname_np_ function in the _OS_Posix_InternalTaskCreate_Impl_ function.  Since this is a non-posix function, each BSP can specify the availability of this function through a #define in osconfig.h

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Dan Burns (GSFC - 596.0)
daniel.s.burns@nasa.gov

astrogeco, 2020-10-01:
@dsburns take a look at #541 that we just merged.

skliper, 2020-10-02:
And I think @jphickey is planning to submit a PR to the PSP to set the thread names using this mechanism.

jphickey, 2020-10-02:
> And I think @jphickey is planning to submit a PR to the PSP to set the thread names using capability implemented in #541.

Done - see nasa/psp#199.

skliper, 2020-12-01:
Closing as duplicate, actually implemented in PSP (nasa/PSP#199)

closed 2020-12-01 by skliper