nasa/osal #542: VxWorks OS_CountSemCreate unit test fails

mbenson1, 2020-07-15:
**Describe the bug**
The OS_CountSemCreate unit test, "#4 Initial-count-too-high", fails.  This is because the vxworks OSAL, as written, does not check the initial count.  Rather, it passes it directly to the OS call.

**Expected behavior**
It should pass.

**System observed on:**
 - AiTech SP0-S 
 - OS: VxWorks 6.9
 - OSAL 5.0.3-bv

**Reporter Info**
Mathew Benson
Windhover Labs

jphickey, 2020-07-15:
Just checked into this - and it is probably related to the value of `SEM_VALUE_MAX`

The test is supposed to be skipped as "NA" if the system doesn't define a specific maximum, but an issue would occur if the `SEM_VALUE_MAX` was actually defined as value 0xFFFFFFFF - in this case it would wrap and therefore pass 0, and the test wouldn't work properly.

jphickey, 2020-07-15:
@mbenson1 can you confirm if your platform defines a `SEM_VALUE_MAX` macro and if so, what is the actual value?

skliper, 2020-12-09:
Pushed a potential fix (#681). @mbenson1 can you confirm this solves the issue?  If not we will need more information to resolve.

PR #681 (2020-12-18): Fix #542, Avoid UT failure if SEM_VALUE_MAX >= UINT32_MAX
**Describe the contribution**
Fix #542 - avoids a potential unit test failure if an invalid semaphore value can not be passed in

**Testing performed**
Build and run unit tests, passed

**Expected behavior changes**
None.

**System(s) tested on**
 - Hardware: cFS Dev Server
 - OS: Ubuntu 18.04
 - Versions: Bundle main + this commit

**Additional context**
Unable to reproduce error from issue (requires feedback from @mbenson1) but this fix avoids a potential error case.

**Third party code**
None

**Contributor Info - All information REQ [... cut]
commit 9a73cb59 (2020-12-18): Fix #542, Avoid UT failure if SEM_VALUE_MAX >= UINT32_MAX
commit d32af54d (2020-12-18): Merge pull request #681 from skliper/fix542-semvaluemax_wrap

Fix #542, Avoid UT failure if SEM_VALUE_MAX >= UINT32_MAX
PR #690 (2020-12-18): OSAL Integration Candidate: 2020-12-22
**Describe the contribution**
Combines IC-2020-12-15 and 2020-12-22

Fix #377 Add "OSAL Select" functional tests
Fix #410, breakup OSAL includes
Fix #542, Avoid UT failure if SEM_VALUE_MAX >= UINT32_MAX
Fix #569, Check and report sysconf error return
Fix #610, Add vxworks 7 support
Fix #670, clean up task definitions in core test
Fix #673 #677, improve global lock on POSIX
Fix #679, Convert OSAL Config Guide to markdown
Fix #686, Use errno in clock_getres error reporting 
Fix #688, Implement macro to facilitate argument checking

**Testing performed**
See [... cut]
closed 2020-12-18 by astrogeco