nasa/osal #543: OS_GetModuleIdByName()

CDKnightNASA, 2020-07-16:
**Is your feature request related to a problem? Please describe.**
If ES loads a module (such as a library that an app uses), but an app wishes to unload that module, that app has no way to find the ID to send to the Unload function.

**Describe the solution you'd like**
Add an OS_GetModuleIdByName() function that returns (via an out param) the ID of the module given its name.

**Requester Info**
Christopher.D.Knight@nasa.gov

jphickey, 2020-07-16:
I am actually somewhat surprised that this isn't already implemented --- probably just an oversight, as every other resource type has such a function and it is simple to implement.  So just for API consistency I concur that this should be added.

BUT ... I'm not entirely sold on the use-case.  nobody should be unloading stuff that was loaded by something else.  ES stores the module IDs of the stuff it loads, so it already knows the ID, and one should call the ES function to unload it.

CDKnightNASA, 2020-07-17:
SBN uses a number of libraries which can be loaded by ES, but if they are not, it will try calling OS_ModuleLoad() itself. Needless to say, it would be nice for SBN to be able to unload them as well.

If nobody should be unloading stuff, then nobody should be loading stuff either. ;)

jphickey, 2020-07-26:
> If nobody should be unloading stuff, then nobody should be loading stuff either. ;)

Possibly true -- but That's a whole separate discussion -- I was mainly referring to the notion of unloading a module that was loaded __by some other app or library__.   An executive (e.g. ES) keeps track of the module ID of everything it loads, and therefore has the capability to unload it using this same ID if/when the time comes.  If an app/lib other than ES loads a module then it can do the same.  But the notion of some other app coming along and unloading something that it didn't load itself seems iffy at best.

CDKnightNASA, 2020-07-27:
ES currently has commands for unloading/reloading applications, perhaps it should have similar for libraries? For that matter, ES should provide a command to load a new application/library that hasn't been configured. I'll file a ticket.

CDKnightNASA, 2020-07-27:
see https://github.com/nasa/cFE/issues/783

I still think GetModuleIdByName() should be supported so I will leave this ticket open.

Should OS_ModuleUnload() have some sort of access control (like only the task that loaded the library can unload it?)