nasa/osal #555: OSAL should provide a typedef for "ID" values

jphickey, 2020-08-11:
**Is your feature request related to a problem? Please describe.**
The OSAL API uses `uint32` type to identify objects/resources in an abstract way.

**Describe the solution you'd like**
There should be a typedef for this.  For instance:

    typedef uint32 OS_id_t;

This typedef can then be used in all APIs rather than using `uint32` directly.

**Describe alternatives you've considered**
Keep using `uint32`.

**Additional context**
Providing a typedef is generally accepted as good practice in terms of future proofing the code.  This wouldn't change anything immediately, as the type will remain `uint32` as it currently is.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2020-08-11:
Ping @skliper , @acudmore --

Stuff I'm working in CFE made me think of this - we are using `uint32` directly in the OSAL API for ID values.  Do you think a typedef is worthwhile to squeeze into the current release before the window closes?  Probably wouldn't be hard to do, just a matter of scrubbing the prototypes/declarations in the API to use the typedef.  Functionality wouldn't change as long as we keep it a `uint32` underneath...

skliper, 2020-08-11:
You mean "Caelum"?  If so, yes, that makes sense to me.

PR #568 (2020-09-02): Fix #555, provide typedef for OSAL ID
**Describe the contribution**
Introduces a new typedef, `osal_id_t`, in `common_types.h`, which should be used to represent an OSAL ID.  All API structures/functions are updated to use this typedef in place of the `uint32` type wherever it actually refers to an OSAL ID.

Adds inline functions for basic equality/validity check, and conversion back to bare integer for logging purposes.

Fixes #555 

**Testing performed**
Build and run all unit tests
Execute and sanity test CFE.

**Expected behavior changes**
Should be none.  As the `osal_id_t` typedef is a ` [... cut]
commit b2a49b17 (2020-09-02): Fix #555, Introduce typedef for OSAL IDs

Update all internal OSAL code to use the typedef when creating, storing,
or otherwise dealing with abstract ID values.

Removes/fixes any code that directly compares or casts the IDs between
normal integer values.  Replace equality comparisons with inline functions
to check for equality and validity.
commit d12808c2 (2020-09-02): Merge pull request #568 from jphickey/fix-555-id-typedef

Fix #555, provide typedef for OSAL ID
PR #590 (2020-09-09): Integration Candidate: 2020-09-02
**Describe the contribution**
Fix #555 
Fix #537 
Fix #581 

**Testing performed**
Bundle CI - https://github.com/nasa/cFS/pull/136/checks

**Expected behavior changes**
PR #568 - Introduces a new `typedef, osal_id_t`, in `common_types.h`, which should be used to represent an OSAL ID. All API structures/functions are updated to use this typedef in place of the uint32 type wherever it actually refers to an OSAL ID.

PR #579 - Fail tests on startup or tear down failures. These are failures, just a different severity (likely mean the test didn't work as expected  [... cut]
closed 2020-09-09 by yammajamma