nasa/osal #556: Deprecate OS_open and OS_creat

jphickey, 2020-08-12:
**Is your feature request related to a problem? Please describe.**
For historical/backward compatibility reasons, the API of these two functions doesn't follow the typical flow.  Rather than providing a `uint32` ID output buffer as the first argument with a separate int32 return code, they return the OSAL ID cast as an `int32` on success.  For these functions, the caller is expected to check if the result is negative, and if so, consider it an error code.  Whereas if it is non-negative, the caller is expected to cast it back to a `uint32` type and interpret it as an OSAL ID.

**Describe the solution you'd like**
These should be like all other OSAL APIs and pass back the ID separately from the return/error status.

**Describe alternatives you've considered**
Leave as is.  But these two functions present a challenge when making a distinct type for OSAL IDs .

**Additional context**
In the current implementation,. these are just compatibility wrappers anyway.  They both call `OS_OpenCreate()` internally, which provides both open (existing file) and creat (new file) based on the flags it was passed.  The `OS_OpenCreate` function _does_ follow the correct pattern so one option would be to just expose this to the public API.

The other option is to create a new version of OS_open and OS_creat which follow the correct pattern.  But in order to provide a transition they would have to use different names.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2020-08-12:
One other possible compromise is to keep these as-is returning `int32`, but provide another function to decode the return value into a status and ID of the correct type (e.g. `uint32`) rather than relying on the user to cast it themselves and do it properly.

This has been a source of confusion in the past, see #498.

jphickey, 2020-08-12:
Would like to get some feedback on this idea in the next CCB.

skliper, 2020-08-12:
I'm in favor of the suggested approach (deprecate, expose OS_OpenCreate).

skliper, 2020-08-12:
@acudmore @ejtimmon ping, deprecation proposal

skliper, 2020-08-12:
CCB 2020-08-12: No objections, but solicited responses/review.

jphickey, 2020-08-12:
Also @CDKnightNASA - you might have some thoughts here as well?

Summary of questions are:

1. Does the inconsistency/confusion around this API warrant making a breaking change to try to fix it and make it like the others?

If (1) is yes, then:

2. Should we keep a separate `open` & `creat` routine or combine them into a single API, using flags to differentiate whether a file should be created or not?  Or should we keep it 1:1 and offer a direct replacement for `OS_open` and `OS_creat`, to make it easier for apps to update?

If (1) is no, then:

3. Is it worth providing a new API to help users check/decode the `int32` value returned from these existing API calls?  This would be backward compatible at least, as it doesn't change/remove any existing function, while alleviating the need for applications to directly cast/convert between OSAL IDs and other integer data types, which is the fundamental concern here.

jwilmot, 2020-08-12:
No objections to just using OS_OpenCreate as this follows the normal API structure found in many file systems where the create option flag is passed in. Hopefully the change only impacts a limited number of applications.

jphickey, 2020-08-12:
Worth noting, in case anyone is unconvinced about the benefits of being type-strict, I found a mistake/bug in the unit tests here, that's probably been lurking for quite some time, where it fails to handle this return code properly:

https://github.com/nasa/osal/blob/8cfd6fe71a5506be8e463f26d92441785fd3e242/src/tests/file-api-test/file-api-test.c#L794-L795

This type of goof-up becomes very apparent/obvious when using a distinct type, and can actually trigger a compiler error if desired.

ejtimmon, 2020-08-12:
> 
> 
> @acudmore @ejtimmon ping, deprecation proposal

This will impact several apps, but I'm not opposed.

skliper, 2020-09-21:
@jphickey - did you start implementing this change, or want me to take it?

jphickey, 2020-09-21:
A previously merged PR has already exposed the `OS_OpenCreate()` API in the public header as discussed, so the next step was to replace current CFE references to `OS_open`/`OS_creat`.

I hadn't started on that apart yet because I needed to get nasa/cfe#868 in first, but now that is settled there shouldn't be any issue in replacing these refs now.  I can submit a CFE ticket for that.  If someone else is available and wants to take it on that would be fine, otherwise I'm happy to do it.

jphickey, 2020-09-21:
See nasa/cfe#893 for the CFE work to be done.

PR #617 (2020-10-13): Fix #556, Deprecate OS_open and OS_creat
**Describe the contribution**

These functions are replaced by OS_OpenCreate, which implements both functions via flags, and follows the correct OSAL API patterns.

Fixes #556 

**Testing performed**
Build and run all tests, sanity check CFE
Check coverage of file-related APIs (still 100%)

**Expected behavior changes**
`OS_open()` and `OS_create()` are deprecated.

**System(s) tested on**
Ubuntu 20.04

**Additional context**
This initially puts the items inside an `OSAL_OMIT_DEPRECATED` block for review and testing, like has been done traditionally.
Sh [... cut]
commit 17767493 (2020-10-13): Fix #556, Deprecate OS_open and OS_creat

These functions are replaced by OS_OpenCreate, which
implements both functions via flags, and follows the
correct OSAL API patterns.
commit 3a76641a (2020-10-13): Merge pull request #617 from jphickey/fix-556-deprecate-open-creat

Fix #556, Deprecate OS_open and OS_creat
PR #621 (2020-10-13): Integration candidate 2020-10-13
**Describe the contribution**
Fix #556
Fix #618

**Testing performed**
See Bundle PR - https://github.com/nasa/cFS/pull/147/checks

**Expected behavior changes**

PR #617 - Deprecate `OS_open` and `OS_creat` to and replaced them with by `OS_OpenCreate`, which implements both functions via flags, and follows the correct OSAL API patterns.

PR #619 - Change use of uint32 for ID to the correct typedef. Also use ObjectIdFromInteger/ObjectIdToInteger where it is intended to convert these values to integers e.g. for the "name" fields in RTEMS.

**System(s) tested on* [... cut]
closed 2020-10-13 by astrogeco