nasa/osal #559: UT_SetForceFail is a misnomer

asgibson, 2020-08-17:
**Describe the bug**
UT_SetForceFail assumes that the value given, that is being set as return value, is a fail condition.  There may be sometimes that a function returns more than 1 value that is not considered a fail.

**To Reproduce**
Use in a unit test where there is a stubbed function with more than 1 "successful" return value.

**Expected behavior**
Change to something like UT_SetForcedReturnValue to make it more generic.  

**Code snips**
N/A

**System observed on:**
RHEL 7.6

**Additional context**
Not a required or debilitating situation, but a name change may be make the function's effects more clear.

**Reporter Info**
Alan Gibson NASA-GSFC/587

jphickey, 2020-08-18:
I concur on that this is not an ideal name --  I was actually thinking this for quite some time but didn't think changing the name was worth breaking test code over it.  But it shouldn't be too hard to apply the deprecation procedure if folks agree to change the name.

How about `UT_SetDefaultReturnValue()`  - I think better indicates the fact that "deferred" values and hook return values will take precedence, and that this value gets used when no other return value is identified.

Should discuss in CCB - fairly easy to do, just need other stakeholders to concur on this one.

asgibson, 2020-08-18:
I agree with your call on the name.

Thanks.

astrogeco, 2020-08-19:
@asgibson will you join the CCB or will @jphickey "represent" you on this?

asgibson, 2020-08-19:
@astrogeco I will not be joining today. I believe @jphickey and I are agreed on the name change. Even if you wish to change the name to something else, as long as the method name is descriptive of what it can accomplish, I am flexible on the actual words used.

astrogeco, 2020-08-19:
[CCB 2020-08-19](https://github.com/nasa/cFS/wiki/Community-CCB-Agenda:-August-19-2020) APPROVED

PR #646 (2020-12-01): Fix #559, Rename set force fail
**Describe the contribution**
Fixes #559 by changing UT_SetForceFail to UT_SetDefaultReturnValue

**Testing performed**
Build and run unit test

**Expected behavior changes**
No impact to behavior

**System(s) tested on**
 Ubuntu 20.04

**Additional context**

**Contributor Info - All information REQUIRED for consideration of pull request**
Alex Campbell - NASA/GSFC
commit 05513686 (2020-12-01): Merge pull request #646 from zanzaben/fix559-Rename-SetForceFail

Fix #559, Rename set force fail
commit 309b2372 (2020-12-01): Fix #559 cleaning up Rebasing
commit 52ee8713 (2020-12-01): fix #559 Refactoring UT_SetForceFail to UT_SetDefaultReturnValue
closed 2020-12-02 by astrogeco