nasa/osal #570: OS_statvfs_t types do not match corresponding POSIX types

klystron78, 2020-08-20:
When building for 64-bit, Wconversion shows issues with OS_statvfs_t's types. They don't match the types in struct statvfs on POSIX systems. It may be worth a discussion to change these types of just handle the conversions in the specific implementation.

jphickey, 2020-08-20:
The `OS_statvfs_t` is currently defined as:

```
typedef struct
{
    uint32 block_size;
    uint64 total_blocks;
    uint64 blocks_free;
} OS_statvfs_t;
```

I assume a conversion warning would have been in relation to the `block_size` element, as it is the only 32-bit value in this struct., and POSIX uses `unsigned long`  Hard to imagine a file system which has a _block size_ that exceeds 32 bits.

FWIW, my experience with `-Wconversion` is that while its interesting and worth looking at the things it complains about, it yields far too many false positives to be useful.  

Would recommend just doing a typecast or something to squelch the warning, rather than change OSAL API.  Unless there is an expectation of seeing a filesystem block size that exceeds 32 bits?

jphickey, 2020-08-20:
On further thought, the struct almost certainly has 32 bits of padding following that value anyway, so changing `block_size` to be a `uint64` type would be pretty benign.  If that fixes the warning, that's fine too - but note it may just shift the warning to later on when an app references this value, so maybe not a fix.

klystron78, 2020-08-20:
I"m fine with not changing it. Put the uint32 at the botttom of the struct to avoid padding . ;)

skliper, 2020-12-09:
Fixed in #654 (confirmed no conversion warning currently on a 64 bit posix build.)

closed 2020-12-09 by skliper