nasa/osal #573: OS_fsBlocksFree

klystron78, 2020-08-20:
OS_fsBlocksFree returns a 32-bit value, but OS_statvfs_t.blocks_free is 64 bit.

jphickey, 2020-08-20:
Similar to #570 -- while this return type (`int32`) is small for modern storages and filesystems, it was seen in previous reviews as being "good enough" for what you'd likely find on a flight system.  Particularly because its returning blocks, not bytes.  It's also overloaded with an error code on failure, which needs to be the `int32` type.

I wouldn't recommend changing the return type for that reason.  Would recommend to deprecate this API rather than change it, if it is really a problem.  New code should just use the existing `OS_statfs()` call anyway, which returns everything.

klystron78, 2020-08-20:
I'm fine with not changing it. I just wanted to bring it to the group's attention.

skliper, 2020-12-09:
There is no OS_statfs.  If OS_fsBlocksFree is removed, seems like there's no replacement (other than using statvfs directly).

skliper, 2020-12-29:
@jphickey could you clarify the "New code should just use the existing `OS_statfs()`" recommendation?

jphickey, 2020-12-29:
Hmm, you are correct, it looks like we do not expose the `OS_FileSysStatVolume_Impl` info via a public API currently.

I can easily implement one as long as folks are in agreement that it is a reasonable solution.  The structure currently is defined as:

```
typedef struct
{
    size_t            block_size;
    osal_blockcount_t total_blocks;
    osal_blockcount_t blocks_free;
} OS_statvfs_t;
```

Prototype would be: 

`int32 OS_FileSysStatVolume(const char *path, OS_statvfs_t *statbuf);`

Filesystem code has traditionally used strings/pathnames to identify objects so this is consistent with other filesystem calls, as opposed to using an osal_id_t to identify the filesystem (which would be more like the rest of OSAL).

jphickey, 2020-12-29:
After this, we would deprecate `OS_fsBlocksFree()` and `OS_fsBytesFree()`.  For the latter, applications would simply multiply `blocks_free * block_size` on their own.

skliper, 2020-12-29:
I'm good with your suggestions above to resolve this issue.

PR #717 (2021-01-12): Fix #573, add OS_FileSysStatVolume
**Describe the contribution**
Add OS_FileSysStatVolume as replacement for OS_fsBytesFree and OS_fsBlocksFree.

Fixes #573

**Testing performed**
Run all unit tests.

**Expected behavior changes**
New API for getting stats on file system.
Uses existing OS_FileSysStatVolume_Impl call - just exposes it in public API.

**System(s) tested on**
Ubuntu 20.04

**Additional context**
This PR is only adding a new API, it is not (yet) deprecating the existing OS_fsBlocksFree/OS_fsBytesFree calls.  Applications and CFE will need an update before these calls can be deprec [... cut]
commit 3fcac5f0 (2021-01-12): Merge pull request #717 from jphickey/fix-573-stat-volume

Fix #573, add OS_FileSysStatVolume
commit b04c4ff8 (2021-01-12): Fix #573, add OS_FileSysStatVolume

Add OS_FileSysStatVolume as replacement for OS_fsBytesFree and
OS_fsBlocksFree.  Update unit tests and stubs for the new API
call.

Does not (yet) deprecate the existing functions, as references
still need to be updated elsewhere in apps.
PR #750 (2021-01-13): osal Integration candidate: 2021-01-12
**Describe the contribution**

Fix #702, use iterators instead of for loops
Fix #645, implement separate lock/unlock key
Fix #703, implement exclusive/reserved lock
Fix #642, make OS_TaskDelete sync
Fix #703, unit test 
Fix #697, use POSIX dir implementation on V
Fix #580, improve FS_BASED mounts on 
Fix #708, chmod error h
Fix #471, order of operations for delete all
Fix #445, add pointer parameter checks
Fix #573, add OS_FileSysSta
Fix #544, add pointer check
Fix #606, Resolve cast-align error in VxWorks OS_TaskGet
Fix #429, update OSAL code to use time [... cut]
closed 2021-01-13 by astrogeco