nasa/osal #577: Subtests no longer executing

skliper, 2020-08-24:
**Describe the bug**
Possibly related to #554 (reverted to v5.1.0-rc1 and they ran), but it doesn't look like subtests are running anymore.

**To Reproduce**
Build with ENABLE_UNIT_TESTS=true, run tests, observe if subtests run (they didn't for msg module).

**Expected behavior**
Subtests should run

**Code snips**
Haven't dug in yet.

**System observed on:**
 - Hardware: cFS Dev Server
 - OS: Ubuntu 18.04
 - Versions checked latest integration candidate and v5.1.0-rc1, problem does not exist in v5.1.0-rc1

**Additional context**
#554 

**Reporter Info**
Jacob Hageman - NASA/GSFC

jphickey, 2020-08-25:
I just checked for SB in the `main` branch and they seem to work there....

skliper, 2020-08-25:
Looks like it's not due to #554.  Try the bundle `integration-candidate`, the `sb` subtests don't look like they execute.

skliper, 2020-08-25:
Starting to look like nasa/cFE#726 may have broken it... still digging in.

skliper, 2020-08-25:
Looks like ut_assert now silently fails to add a test from inside another test.  This worked pre #554.  `sb` didn't do this, but `msg` did.

skliper, 2020-08-25:
Or I should probably say doesn't add the test in a way that it will get executed, since it gets added to the front of the list (at least that's what it looks like on quick inspection.)

jphickey, 2020-08-31:
Ahh yes, that would explain it - because that happens after combining the setup+test+teardown lists into one summary list, the newly-added test is still sitting on one of the component lists, not the summary list.

If this is a desired/necessary feature (to add tests during tests) then perhaps we can add an outer loop into UtTest_Run to pick up tests that been added during the run.  Would be simple to add.

skliper, 2020-12-29:
@astrogeco - can we close this?  No requirement or real use case identified... just don't add test cases from other test cases.  No need to implement additional logic simply to add complexity.

PR #751 (2021-01-21): Fix #577, Document nested tests not supported
**Describe the contribution**
Fix #577 - adds documentation to the UtTest_Add API that nesting (UtTest_Add from within an added test) silently fails.

**Testing performed**
None, documentation only.  

**Expected behavior changes**
None

**System(s) tested on**
N/A

**Additional context**
nasa/cFE#841

**Third party code**
None

**Contributor Info - All information REQUIRED for consideration of pull request**
Jacob Hageman - NASA/GSFC
commit 62eb12d5 (2021-01-21): Fix #577, Document nested tests not supported
commit 86b13a2a (2021-01-21): Merge pull request #751 from skliper/fix577-nested_add

Fix #577, Document nested tests not supported
PR #767 (2021-01-27): osal Integration candidate: 2021-01-19
**Describe the contribution**

Fix #746, Add UtDebug message from OS_printf stub
Fix #692, Resolve typos in unit test assert OR statements
Fix #577, Document nested tests not supported
Fix #753, Remove UT_CheckForOpenSockets
Fix #229, add mqueue functional test
Fix #757, Readd extern C in osapi.h to support C++ use
Fix #692, display message when not implemented error occurs
Fix #591, add test teardown failure to test summary
Fix #699, format printf correctly
Fix #685, change file create to use read write
Fix #748, add additional casts
Fix #743, Create Sec [... cut]
closed 2021-01-27 by astrogeco