nasa/osal #578: Clarify usage of UT_ADD_TEST and clarify nomenclature for individual versus groups of tests

astrogeco, 2020-08-25:
Related to nasa/osal#577.  Individual tests are added w/ the UT_ADD_TEST call.  Some tests are grouped at a lower level (Test_MSG_CCSDSExt is a group of tests), and those functions are called directly.  Same pattern is used in `sb`:

https://github.com/nasa/cFE/blob/a148b97511de3f45f6677c1a3a0b17197f867e28/fsw/cfe-core/unit-test/sb_UT.c#L134-L135

_Originally posted by @skliper in https://github.com/nasa/cFE/pull/840#issuecomment-680193849_

skliper, 2020-08-25:
This may belong in nasa/cfe, which own `UT_ADD_TEST` and `UT_AddSubTest`, and the unit tests that employ this model.

astrogeco, 2020-08-25:
> This may belong in nasa/cfe, which own `UT_ADD_TEST` and `UT_AddSubTest`, and the unit tests that employ this model.

ah I put it here since you mentioned UT_Assert :(

skliper, 2020-08-25:
The `osal` relation is you shouldn't call `UtTest_Add` from within another test that was registered with `UtTest_Add`.

astrogeco, 2020-08-25:
Would it make sense to keep both of them open then? maybe reframe #577 as prevent that behavior or document it?

astrogeco, 2020-08-25:
Edit: move this one to cfe but reopen and reframe #577

skliper, 2020-08-25:
I'm fine with however you want to handle it.

astrogeco, 2020-08-25:
Moved to nasa/cfe#841

closed 2020-08-25 by astrogeco