nasa/osal #588: Add tracking for order of stub calls during a unit test

asgibson, 2020-09-02:
**Is your feature request related to a problem? Please describe.**
It can sometimes be important to know that calls in a function are done in a proper order; i.e. ability to verify that a message is timestamped before it is sent.

**Describe the solution you'd like**
An addition that allows a stub to report that it was called to a central location and this information accessible to a unit test for verification.  Add this to the default stub implementation, but have it be available to stubs that do not use the default.

**Describe alternatives you've considered**
Write local hooks for all stubs used in unit testing that use a local version of this concept.

**Additional context**
Example of situation that requires this:
```    
CFE_SB_TimeStampMsg((CFE_SB_Msg_t*) &XX_AppData.hk);   
CFE_SB_SendMsg((CFE_SB_Msg_t*) &XX_AppData.hk);
```

**Requester Info**
Alan Gibson NASA GSFC/587

skliper, 2020-09-02:
Is there an opportunity to avoid additional complexity by separating the concepts of a coverage test and a functional test?  I'd expect the coverage test (with mocks and stubs) to focus on coverage, and the functional test (utilizing the real stack) to confirm the functionality (such as the message getting timestamped correctly).  Either way the functional test should confirm this behavior, so it doesn't seem like a coverage test adds a lot.  Basically the fact APIs get called in a certain order doesn't really prove the functionality... only proving the functionality works as expected meets that requirement.

skliper, 2020-09-02:
Checking the order of API calls from a coverage test only shows that the order matches what you expect in the code under test.  That order could functionally meet the requirement or not (you don't know from a coverage test, since it's all stubs).  You won't know that until you do a functional test.

asgibson, 2020-09-02:
> Is there an opportunity to avoid additional complexity by separating the concepts of a coverage test and a functional test? I'd expect the coverage test (with mocks and stubs) to focus on coverage, and the functional test (utilizing the real stack) to confirm the functionality (such as the message getting timestamped correctly). Either way the functional test should confirm this behavior, so it doesn't seem like a coverage test checking the order of API calls adds a lot. Basically the fact APIs get called in a certain order doesn't really prove the functionality... only proving the functionality works as expected meets that requirement.

I must admit, with the term ut_assert, I thought the name implies that it is a "unit testing framework".  My definition of a unit test is testing the smallest amount of code possible (I have considered adopting the term 'method level testing' due to conflicting definitions).  I expected that ut_assert will only be used for that because of the name.  I see the potential to use it as a more expansive functional testing framework, but it does not currently seem geared towards that even while it can be used.

Functional tests will confirm this behavior, but why not ensure that down at the unit test level?  This is especially true when doing test driven development.  My unit test says that this must be the true and then when I develop the code (after writing the test and seeing it fail), I ensure that I have written the code correctly.  The functional test will then pass when it is written and, instead of finding this problem later during functional testsing, we have made it never occur.  There are situations, as the developer of the code, that I know the ordering that is required regardless of whether or not there is a functional requirement for it above my level of development.

> Checking the order of API calls from a coverage test **_only shows that the order matches what you expect in the code under test_**. That order could functionally meet the requirement or not (you [...]

This is exactly what I seek to verify.

skliper, 2020-09-02:
If you find value in checking that APIs are called in a certain order as part of your testing I suggest a hook.  It's not general practice in our coverage tests since checking the order of API calls does not satisfy one of our unit test requirements or verify functionality so it's not considered "high value" within the context of our coverage tests.

asgibson, 2020-09-02:
Can you point me to the unit test requirements?

asgibson, 2020-09-02:
Thanks @skliper for the email about unit test requirements.

asgibson, 2020-09-02:
If the requirement of my function (which in this case is synonymous with the 'code under test') is to timestamp a message and then send it out, why does it not make sense to test that the timestamp was done first?  It can be checked that timestamp was called and that the message was sent, but that doesn't seem to fully check that the function fulfilled its obligations.

A unit test that only checks that they were called 'passes' on both:
```
CFE_SB_TimeStampMsg((CFE_SB_Msg_t*) &XX_AppData.hk);   
CFE_SB_SendMsg((CFE_SB_Msg_t*) &XX_AppData.hk);
```
and
```
CFE_SB_SendMsg((CFE_SB_Msg_t*) &XX_AppData.hk);
CFE_SB_TimeStampMsg((CFE_SB_Msg_t*) &XX_AppData.hk);
```

jphickey, 2020-09-02:
That strikes me more as a functional test objective, because the two functions (CFE_SB_SendMsg and CFE_SB_TimeStampMsg) are stubs that don't actually touch the message.  Coverage test should just check that the path was followed.  Only a functional test can confirm that the timestamp was actually set correctly within the message.

Coverage tests can only check functionality that is locally implemented within the same unit, not external items.  Nor is it a specific requirement to invoke a particular API.  For instance one could call `CFE_SB_SendMsgFull` or even `CFE_SB_ZeroCopySend` which all provide different flavors of the same function (sending a message) and checking for a specific one doesn't really add test value (IMO) but does become a maintenance issue.

skliper, 2020-09-02:
Calling two APIs in order does not prove functionality, and I wouldn't accept it as verification of a functional requirement.  For example the real implementation of CFE_SB_TimeStampMsg may have specific restrictions on message type that are not met by your unit, and it will still functionally not meet the requirement even though the unit calls the two APIs in the right order.  It's just not what I'd consider "high-value" in a coverage test.

skliper, 2020-09-02:
> I must admit, with the term ut_assert, I thought the name implies that it is a "unit testing framework". My definition of a unit test is testing the smallest amount of code possible (I have [...]

@asgibson - we are currently using ut_assert at a coverage level (with mocks/stubs), in stand alone **functional** tests for OSAL (real stack), and in an application context within cFS for functional testing of cFE and PSP APIs.  It's definitely flexible/adaptable to may different forms and levels of "unit testing".  It's just compiled within different contexts.

asgibson, 2020-09-02:
I suppose this is where we all differ in our definition of unit testing.  I am testing code behavior. I seek to ensure that the code behaves as expected, so that when something else uses the code, it does what it is intended to do.  This is not functional level testing nor is it integration level testing, this is what I see as unit testing.  I had not heard the term _coverage testing_ until I came to NASA. 

I do realize that the term **_unit_** can apply to any level of testing because **_unit_** is an ambiguous term. I will turn to the preeminent authority Wikipedia ;-) where the last sentence of the first paragraph of the unit testing page states, "By writing tests first for the smallest testable units, then the compound behaviors between those, one can build up comprehensive tests for complex applications."  When I say unit testing, I am talking about testing the smallest testable units and I am talking about writing these tests first.

My view of the levels of testing: unit test -> integration test -> functional test -> performance test -> acceptance test 

Having 100% code coverage does not ensure: proper testing, that all code behavior is tested or that anything has been verified at all. Comment out just the asserts from a test suite and run them; test coverage does not change and no behavior has been verified (which coincidentally at a top level `make test` run shows absolutely no difference in output, https://github.com/nasa/cFE/issues/855, because there is no output of the individual test's assert fail/total ratio).  The only thing I know is that my tests are setup properly so as to not segfault while running.

Waiting until functional testing to discover problems in the code that could have been seen during the testing of the smallest units is wasting time and effort to go back, find and fix an issue that could have been avoided early.

skliper, 2020-09-03:
> I suppose this is where we all differ in our definition of unit testing...

Which is why I provided you our internal requirements that satisfy our criteria for "unit testing" and our mapping of testing activities to those requirements.  Note I have not said coverage testing satisfies all those requirements.  Note I have also not said you need to wait to do functional testing.  I often write those test as I'm developing.  How do you know your code works if you don't functionally test it?  The point is it's not possible to meet all the unit test requirements within just one context (just compiled with mocks/stubs, or just compiled w/ the full stack), so efficiencies can be realized by focusing on the requirements you can satisfy in each context vs trying to do everything in one test.  That said, feel free to do whatever you want, just stating our approach.

My offer remains open to discuss in real time to avoid endless threads and potentially address misunderstandings up front.

skliper, 2020-09-03:
So where does your comment "waiting for functional test" comment come from then, if you are functionally testing at a unit level?  I get the feeling we are just saying the same thing in two different ways.  "Unit test" requirements include code coverage and functional testing.  We accomplish this via two different contexts, since you can't get full coverage with the full stack (can't force errors), and can't verify functionality without the full stack.

Wrapping back to the initial request - the cFS framework does not use order of API calls in a coverage test (with mocks and stubs) to verify functionality.  We use a functional tests (at whatever level is appropriate to verify the function) to verify function.  Again, you are welcome to do whatever you want in your testing, these are just suggestions for how you may be able to meet the unit test requirements without requiring this requested capability.

asgibson, 2020-09-03:
At the smallest unit test level, I am verifying method behaviors.  What do coverage tests seek to verify?

jphickey, 2020-09-03:
> At the smallest unit test level, I am verifying method behaviors

This is what I've been trying to say - that in this mock/stub environment you can only verify behavior of something that is _entirely_ implemented within the unit under test and does not invoke any dependent functions (which are stubs).  Because the behavior of the stub is incorrect (by definition) and therefore makes the dependent behavior also incorrect (by definition).

So sure - for functions that are entirely within the unit, and do not have any dependencies - it is fine to verify the behavior of these methods and this environment may be a useful place to implement that.

But for anything which does call any non-local function, verifying behavior in this environment is a flawed paradigm.

> What do coverage tests seek to verify?

In my opinion - coverage test value is where it can check for undefined/invalid code in hard-to-reach error handling cases.  

For example, if a certain path through the code yields a local pointer to be NULL and later code  tries to de-reference that pointer.... coverage testing that exercises all paths can catch this.

Obviously this isn't perfect either - sometimes only a certain _combination_ of branches produces an error - which means testing every combination would be required - which is not feasible in any non-trivial function.  But this is more of a code design/architecture issue - it is why individual functions should be kept simple as possible.  Many areas of CFE - TBL code in particular - are pretty bad here - with functions hundreds of lines long and many nested branches.

I do think that historically too much weight/effort has been invested in the coverage test environment.  Primary focus should really be on full stack/functional test first.  Then mock/stub coverage to fill the gaps.

asgibson, 2020-09-03:
> This is what I've been trying to say - that in this mock/stub environment you can only verify behavior of something that is entirely implemented within the unit under test and does not invoke any [...]

I do not think this is true.  I can verify "expected" behavior of the method under test.  Integration testing verifies that more than 1 "unit" functions together properly.  If I am designing a method, I am assuming that I understand how the things I am using work, my tests reflect this assumption.  After all, if I don't know how things are supposed to work, how can my method design be correct in the first place?  If that assumption is incorrect, my design is wrong and the method level test will not see this, it is not designed to do that!  The integration test will locate these, which it is designed to do.  It's all about targeted verification results, so less time is spent tracking down the problem.

>I do think that historically too much weight/effort has been invested in the coverage test environment. Primary focus should really be on full stack/functional test first. Then mock/stub coverage to fill the gaps.

On this we definitively disagree.  Methods built by test driven design tend to be better designed functions, because when something is difficult to test, questions arise about the design.  TDD encourages better design and prevents things like your observation below:
>Many areas of CFE - TBL code in particular - are pretty bad here - with functions hundreds of lines long and many nested branches.

jphickey, 2020-09-03:
> I do not think this is true

I can't see how one can argue differently here.  To use another analogy, lets say the unit does some addition and relies on the fact that `2+2 = 4`.  Lets say (hypothetically) under unit test I replace the addition logic with a stub which can produce any result I wish, and now `2+2=10`.

Now the unit under test is going to get into all sorts of incorrect behavior beyond this, because it is using an incorrect sum, and so on and so forth.  It might update the wrong table entry, causing subsequent logic to do the wrong thing, or any number of things.  The point is, all bets are off when using a stub that doesn't behave according to its API contract.

You might think that this is a far-fetched scenario because arithmetic log is built into the CPU, but I don't think that matters.  The concept is the same.  Consider C library functions like `memset()` -- what if this was a stub that didn't actually clear the memory, and then code under test referenced that memory later?  Undefined behavior.  These are obviously simple examples, but the same problem exists at every level.   Units depend on OSAL and CFE calls performing the operation that their API says they do, and if they don't then the entire behavior becomes basically undefined.

asgibson, 2020-09-03:
@jphickey I would like to respond to that and I value this conversation, but we are way off the issue topic.  Is there a better place to continue this broader discussion of testing?
https://github.com/nasa/cFE/issues/855 has landed on much the same subject.

I was also informed recently, by a trusted colleague of mine, that my earlier small example was not doing the complexities of testing justice.  If we are truly going to delve into this, perhaps we should be working with real world examples.

skliper, 2021-01-11:
Note that all the stub calls are reported in order of execution when you test with debug enabled (-d).  I'm not sure what a built in order verification API would look like, but you can always post-process the debug log if order verification is required.  It's also possible via custom hooks to check order.  Closing this issue based on those options being available to verify order of execution.  Feel free to re-open if this is not sufficient or there is a suggested solution/implementation you'd prefer.

closed 2021-01-11 by skliper