nasa/osal #608: Add support for RTEMS 5.1 - RTEMS OSAL and pc-rtems OSAL BSP updates

acudmore, 2020-09-25:
**Is your feature request related to a problem? Please describe.**
The cFS Bundle currently supports RTEMS 4.11. Now that RTEMS 5.1 has been released, I would like to update the necessary components to support RTEMS 5.1 on the pc-rtems platform. This involves minor modifications to the cfe repository, the PSP repository, and the OSAL repository. These changes can be done in such a way that preserves the current RTEMS 4.11 support and adds RTEMS 5.1 support.

**Describe the solution you'd like**
I would like to add support for RTEMS 5.1 while maintaining compatibility with the existing RTEMS 4.11 support. For the OSAL repository there are a few minor changes needed to allow support of RTEMS 5.1 and RTEMS 4.11:
- Minor changes in the OSAL pc-rtems BSP
- Minor changes in the RTEMS OSAL implementation files

**Describe alternatives you've considered**
Alternatives include not supporting newer versions of RTEMS, but several projects will depend on RTEMS 5 support.

**Additional context**

**Requester Info**
Alan Cudmore / NASA GSFC Code 582

joelsherrill, 2020-09-25:
I think I know of one project that used cFS and RTEMS 4.11. Obviously, there could be many others. Whether any cFS users ever update RTEMS or cFS might be an interesting topic. I suspect they select and freeze with no upgrade. The next project may move to a newer version.

If projects are using SMP, I would recommend they move to 5.x.

acudmore, 2020-09-25:
Related to:
CFE 914: https://github.com/nasa/cFE/issues/914
PSP 197: https://github.com/nasa/PSP/issues/197

PR #661 (2020-12-07): Fix #608 - Add RTEMS 5.x support
**Describe the contribution**
Fix #608

This change adds support for RTEMS 5.1 in the OSAL.
The change provides defines and necessary ifdefs so RTEMS 4.11 can continue to be supported.

**Testing performed**
Following the instructions in the pc-rtems PSP:
https://github.com/nasa/PSP/blob/main/fsw/pc-rtems/README.txt
and the updated README_RTEMS_5.txt from PSP pull request 197, I built the RTEMS toolchains, the pc686 BSPs and cFS bundle for both RTEMS 4.11 and RTEMS 5.x. I then ran the bundle on both RTEMS 5.1 and RTEMS 4.11 on QEMU x86.
Note, that for RTEMS 5,  [... cut]
PR #680 (2020-12-09): OSAL Integration Candidate: 2020-12-08
**Describe the contribution**

Fix #120, Exit console loop on shutdown
Fix #290, Remove obsolete printf tests
Fix #608, Add RTEMS 5.x support
Fix #631, add new functional tests for OS_chmod
Fix #648, scrub array references in shared layer
Fix #648, rename internal fields for consistency
Fix #648, sanitize array refs on impl layer
Fix #649, use full object ID in timecb list
Fix #650, OS_chmod uses read or write access.
Fix #664, change type of sync callback argument to osal_id_t

**Expected behavior changes**

PR #661 - Add support for RTEMS 5.1. The chang [... cut]
commit 962bd539 (2020-12-09): Merge pull request #661 from acudmore/fix608-add-rtems5-support

Fix #608 - Add RTEMS 5.x support
commit b7680650 (2020-12-09): Fix #608 - Add RTEMS 5.x support
closed 2020-12-09 by astrogeco