nasa/osal #620: FreeRTOS port

jonathanmichel, 2020-10-07:
Hi,

It seems that you only provide the OSAL for Linux, rtems and vxworks. However you regularly mention FreeRTOS in your documentation. Do you plan to share the sources for the FreeRTOS port ? Not sure the amount of work such operation require and the limitations (especially concerning the file system) . It would be great to have more information about this point. 

Thanks in advance !

**Requester Info**
Jonathan Michel, master student at the University of Applied Science of Western Switzerland

skliper, 2020-10-07:
Hi Jonathan - the best resource for OSAL implementations that are not provided as part of the cFS Framework is the community email list.  FreeRTOS is occasionally a topic on that list, and there are a few related community efforts that are likely much more up-to-date (as of last year it looks like they were targeting FreeRTOS 10) than anything I have access to.  The subscription request can be found here: https://lists.nasa.gov/mailman/listinfo/cfs-community.

closed 2020-10-07 by skliper

skliper, 2020-10-07:
Once you are in, the most recent FreeRTOS thread I found was: https://lists.nasa.gov/mailman/private/cfs-community/2019-December/001399.html (access for community members only).

jonathanmichel, 2020-10-07:
Thanks I will take a look. I subscribed to the community and am waiting for the confirmation for now.

jonathanmichel, 2020-10-08:
@skliper Is the [email list website](https://lists.nasa.gov/mailman/listinfo/cfs-community) down on your side too ?

skliper, 2020-10-08:
@jonathanmichel Just tested and I am able to sign in and get to the archives

jonathanmichel, 2020-10-08:
@skliper Strange. It was working yesterday but I never received the email confirmation. And today the website is down (ERR_CONNECTION_TIMED_OUT on my computer and ERR_CONNECTION_TIMED_RESET on my mobile phone). Do you have any idea ?

skliper, 2020-10-08:
@jonathanmichel contact me via email and I'll see what I can do.  Any pull request I have submitted will have a commit w/ my email in it.

aogrcs, 2020-11-24:
@skliper 
Dear,
I found that I could not open this site, thanks

skliper, 2020-11-24:
@aogrcs - you need to join the community email list at https://lists.nasa.gov/mailman/listinfo/cfs-community prior to being able to view the archives.

aogrcs, 2020-11-25:
@skliper,
Is the website down for country outside US? Thanks

skliper, 2020-11-25:
@aogrcs - good question.  If the links above for the community email list don't work, you could try the contact information associated with top results on a search (https://www.google.com/search?q=cfs+freertos).  FreeRTOS isn't officially supported from nasa/OSAL, but there are ports the wider community have developed.

aogrcs, 2020-11-26:
@skliper maybe I should do it myself, thanks

PratikAher20, 2022-11-04:
> @skliper Strange. It was working yesterday but I never received the email confirmation. And today the website is down (ERR_CONNECTION_TIMED_OUT on my computer and ERR_CONNECTION_TIMED_RESET on my mobile phone). Do you have any idea ?
Facing the same problem... Please Help

skliper, 2022-11-07:
Hi @PratikAher20, @dzbaker @dmknutsen are the prime contacts these days for cFS.