nasa/osal #622: OS_FileRename_Impl yields different results on different operating systems

klystron78, 2020-10-15:
**Describe the bug**
An FM test is showing different behavior on Linux and vxWorks. The issue is in a test that does a file rename. There's a test that renames a file to the name of an existing file. On Linux, this file is overwritten. On vxWorks, the call to rename() fails. It would be good to have consistent behavior for FM across both operating systems.

**To Reproduce**
Walt Moleski can provide more information on reproducing.

**Expected behavior**
It would be ok for the rename to fail or for the rename to succeed. The CCB should decide what the desired behavior should be. FM test except the rename to succeed.

The expected result is consistency across both operating systems, so that FM can pass its tests on both.

**System observed on:**
Linux (PC) and vxWorks 6.9 (mcp750?)

**Reporter Info**
Steven Seeger GSFC - 582 (Embedded Flight Systems, Inc.)

jphickey, 2020-10-15:
Hmm, this is a common paradigm to use `rename` to atomically overwrite an existing file.  (i.e. from the point of view of another task - they'll either get the old file or the new file - there will be no moment in time where opening the file might fail because the file doesn't exist).  POSIX actually describes this atomic overwrite behavior in their spec for `rename()`.

It sounds like the underlying VxWorks file system is lacking here - although they've adhered to the POSIX API for rename the behavior is non-compliant.

The only possible thing we could do in OSAL is to remove the destination file first.  It wouldn't be atomic - so it becomes a race condition.

The question then becomes - is having a race condition on VxWorks better than the current inconsistent behavior?

skliper, 2020-10-15:
Does VxWorks 7 do it the right way?

jphickey, 2020-10-16:
> Does VxWorks 7 do it the right way?

If VxWorks is like most other OS's that have multiple filesystems, ops like this are generally part of the file system not the core kernel.

I am wondering what file system was in use when this behavior was observed, and if perhaps using a different file system yields different behavior?

klystron78, 2020-10-21:
> Does VxWorks 7 do it the right way?

I missed the comment here. I can boot my vxWorks7  box and test rename and answer your question later today.

jphickey, 2020-10-21:
A different but related item -- while you are checking `rename()`, can you also check what happens when you call `OS_OpenCreate()` or `open()` with TRUNCATE flag on an existing file?  

I see code in CFE like this: https://github.com/nasa/cFE/blob/dc3d62b37d2bacd8fe23f66ff7b64025ac7d0751/fsw/cfe-core/src/es/cfe_es_task.c#L1193-L1208

I'll probably write a CFE issue about it, but this "Check to see if the file already exists" should be entirely unnecessary.  Unless someone was working around a non-compliant file system implementation that couldn't truncate an existing file during `open()`.

klystron78, 2020-10-21:
> Does VxWorks 7 do it the right way?

fails on vxWorks 7

-rwxrwxrwx  1 0       0               900096 Jan  1  1980 blah 
-rwxrwxrwx  1 0       0             32505856 Jan  1  1980 blah2 
value = 0 = 0x0
-> rename "blah2", "blah"
value = -1 = 0xffffffffffffffff

klystron78, 2020-10-21:
> A different but related item -- while you are checking `rename()`, can you also check what happens when you call `OS_OpenCreate()` or `open()` with TRUNCATE flag on an existing file?
>
> I see code in [...]

I just did at the shell open "blah", 0x400

And I got a valid fd (4) and "blah" went from 900096 bytes to 0 bytes.

This is on a dosFs ramdisk. (as was my test with rename)

astrogeco, 2020-10-28:
[CCB 2020-10-28](https://github.com/nasa/cFS/wiki/Community-CCB-Agenda:-October-28-2020) 
- Spurred by FM tests using the rename function
- Related to previous discussion on modifying open files, conclusion at that point was to defer to underlying OS behavior
- In this situation we should try to make the behavior as close as possible between operating systems
- There is potential for race conditions in posix OSs since there's a window where the "file is not there"
- Do/should apps rename files?
- Should the rename operation be up to operators
- Need @ejtimmon to weigh in on conversation
- Scheduling splinter

skliper, 2020-12-01:
I think we decided to defer to underlying OS behavior (like delete when open, etc).  The app change that triggered this issue has been backed out, so closing as "won't fix".  Feel free to reopen if needed.

closed 2020-12-01 by skliper