nasa/osal #635: More consistent types for OSAL resources

jphickey, 2020-10-29:
**Is your feature request related to a problem? Please describe.**
CFE issue nasa/cfe#663 describes mismatched types used across CFE for certain items, such as stack size and priority of tasks.
To help resolve this OSAL should provide a typedef for the "right" types to use when interfacing with OSAL.

**Describe the solution you'd like**
Should add at least:
- `osal_priority_t` - type used for OSAL task priority
- `osal_stackptr_t` - type used for OSAL stack pointer

**Additional context**
This will identify the correct type to use, instead of the mishmash currently seen in CFE as described in nasa/cfe#663.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2020-10-29:
Question for @skliper and @acudmore --- OSAL also (still) uses `uint32` for size everywhere.  We already have a typedef for `cpusize` aka `size_t` - should I also update the OSAL APIs to use `cpusize` instead of `uint32` in the APIs that accept a size?

jphickey, 2020-10-29:
Otherwise I can limit this to only fixing the type used for priority and the stack pointer (which only occur in a couple places).

skliper, 2020-10-29:
I'm good with the size change also.

PR #654 (2020-11-21): Fix #635, Reduce use of uint32, add more OSAL typedefs
**Describe the contribution**

Scrub all uses of `uint32` type across OSAL, and replace most usage of this generic type with more appropriate, purpose-specific typedefs.

All use of `uint32` to store an object size is now replaced with `size_t`.

Adds proper typedefs for:
    
- `osal_priority_t` - used for indicating a task priority
- `osal_stackptr_t` - for indicating a task stack pointer
- `osal_index_t` - for indicating a table position/array index of all internal tables
- `osal_objtype_t` - for indicating the type of an object
    
Fi [... cut]
commit 2899dfec (2020-11-21): Fix #635, add typedefs for osal stack and priority

Adds typedefs for:

- osal_priority_t
- osal_stackptr_t
- osal_index_t
- osal_objtype_t
- osal_blockcount_t

Note that by using `uint8` as the priority type, all values are
valid and it is no longer possible to pass a value which is out
of range.  Checks/tests for this are no longer valid and would
cause compiler warnings.
commit 5db9bdbe (2020-11-21): Merge pull request #654 from jphickey/fix-635-osal-add-task-typedefs

Fix #635, Reduce use of uint32, add more OSAL typedefs
PR #662 (2020-12-02): osal Integration Candidate: 2020-11-24
**Describe the contribution**

Fix #559, Refactoring UT_SetForceFail to UT_SetDefaultReturnValue
Fix #635, add typedefs for osal stack and priority
Fix #655, use 3 argument form of open
Fix #651, use unsigned types in VxWorks stack calculation
Fix #559, cleaning up Rebasing
Fix #586, add functional test of one shot timer

**Testing performed**

**Expected behavior changes**

PR #646 - Rename `UT_SetForceFail` to `UT_SetDefaultReturnValue` since some functions that retain more than 1 value are not necessarily failing

PR #653 - Add a 5th timer to TimerTest [... cut]
closed 2020-12-02 by astrogeco