nasa/osal #650: OS_chmod is requiring read access

zanzaben, 2020-11-10:
**Describe the bug**
If you don't have read access to a file then OS_chmod will not be able to change it. This is most likely caused by chmod opening the file as a way to avoid filename race potential, see the code snip. 

**To Reproduce**
1) Set the file access to write only. 
2) Try to change the access using OS_chmod
3) will get an OS_ERROR

**Expected behavior**
You should be able to change the permissions of a file without read access. 

**Code snips**
https://github.com/nasa/osal/blob/1a826576f8dd9eb37f47da0e633b1d4712b8ac61/src/os/portable/os-impl-posix-files.c#L201-L206

**System observed on:**
 Ubuntu 20.04

**Reporter Info**
Alex Campbell GSFC

jphickey, 2020-11-10:
I think this came up in a previous CCB at one point, when it was changed to use `fstat` / `fchmod` to do this on a file descriptor rather than file name....

Linux offers a solution in via the `O_PATH` flag - which opens the fd in a special mode where normal reading/writing is not possible but stat/chmod are possible.  But AFAIK this flag is a non-posix extension.  Perhaps it warrants some sort of `#ifdef` logic to determine of the OS provides this extension and use it if it does.

astrogeco, 2020-11-12:
[CCB 2020-11-12](https://github.com/nasa/cFS/wiki/Community-CCB-Agenda:-November-12-2020) 

We discussed this a while ago when we talked about 
Depends on the file system implementation
No easy way to fix this in POSIX
DOS-FS does not use this so we don't know what happens
Add an `ifdef` and use `O_PATH` if it exists

jphickey, 2020-11-12:
Regarding the actual test cases -- my suggestion is to still run through the test, but make the one where it tries to re-enable read access to be an "MIR" type if it fails.  

That is -- if it succeeds and returns OS_SUCCESS then the test passes no problem - but if it fails with non-OS_SUCCESS, then report it as MIR - so it will not stop the test.  

This way if the system does support `O_PATH` we can actually exercise its intended purpose and confirm its working as expected, but on a system that doesn't have `O_PATH` the test case is just skipped.  (MIR doesn't fail the test).

zanzaben, 2020-11-18:
If it is not able to be opened in read only can we then try to open it in write only? All that matters is that the file is open to prevent race potential.

skliper, 2020-11-30:
> If it is not able to be opened in read only can we then try to open it in write only? All that matters is that the file is open to prevent race potential.

After a chance to think about this, seems preferable to the system-dependent `O_PATH` approach.  Could you submit a PR?  Since the only options are `OS_READ_ONLY`, `OS_WRITE_ONLY`, and `OS_READ_WRITE` trying both one should succeed.

jphickey, 2020-12-01:
> After a chance to think about this, seems preferable to the system-dependent `O_PATH` approach. Could you submit a PR? Since the only options are `OS_READ_ONLY`, `OS_WRITE_ONLY`, and `OS_READ_WRITE` [...]

Note that it is possible to have neither read nor write permission on a file, even for the file's owner (i.e. mode `0000`).  The `O_PATH` still works even for this, but opening for write will not.

skliper, 2020-12-01:
> ...possible to have neither

Understood (which is why I was hesitant at first w/ the suggestion), but it seems like the OSAL only supports setting it as the listed three?  If OSAL doesn't support setting it as neither, does OS_chmod really need to support it conditionally based on if `O_PATH` is available?  Seems like added complexity that doesn't universally solve the problem, so I'm leaning towards consistency...

PR #663 (2020-12-07): fix #650 OS_chmod uses read or write access.
**Describe the contribution**
Fixes #650 by checking for either read or write access instead of just read access 

**Testing performed**
Used the Chmod tests I am building in another branch that found this issue in the first place. 

**Expected behavior changes**
A clear and concise description of how this contribution will change behavior and level of impact.
-No changes 

**System(s) tested on**
Ubuntu 20.04

**Contributor Info - All information REQUIRED for consideration of pull request**
Alex Campbell GSFC
commit 0ecfba45 (2020-12-07): fix #650 OS_chmod uses read or write access.
commit 77300fda (2020-12-07): Merge pull request #663 from zanzaben/fix650_oschmod_access

Fix #650, OS_chmod uses read or write access.
PR #680 (2020-12-09): OSAL Integration Candidate: 2020-12-08
**Describe the contribution**

Fix #120, Exit console loop on shutdown
Fix #290, Remove obsolete printf tests
Fix #608, Add RTEMS 5.x support
Fix #631, add new functional tests for OS_chmod
Fix #648, scrub array references in shared layer
Fix #648, rename internal fields for consistency
Fix #648, sanitize array refs on impl layer
Fix #649, use full object ID in timecb list
Fix #650, OS_chmod uses read or write access.
Fix #664, change type of sync callback argument to osal_id_t

**Expected behavior changes**

PR #661 - Add support for RTEMS 5.1. The chang [... cut]
closed 2020-12-09 by astrogeco