nasa/osal #672: Add a stand-alone example build with osal (modernize src/examples/ tasking-example?)

skliper, 2020-12-08:
**Is your feature request related to a problem? Please describe.**
Difficult for new user to set up a new trivial build using just OSAL.

**Describe the solution you'd like**
Helpful to include a simple CMake setup and a bare-bones c file to show simplest version of OSAL use.  There's src/examples, but it's not set up to build/run out of the box.

**Describe alternatives you've considered**
None.

**Additional context**
None.

**Requester Info**
Jacob Hageman - NASA/GSFC - Stakeholder request

astrogeco, 2021-05-17:
@skliper Does https://github.com/nasa/osal/pull/812 fix this?

skliper, 2021-05-17:
> @skliper Does #812 fix this?

No.  #812 does supply the sample cmake instructions, but this is to actually add something that will build/run.  Even better if the example build is added to CI so it doesn't rot.