nasa/osal #685: File chmod test failing on RTEMS

jphickey, 2020-12-11:
**Describe the bug**
When running unit tests against the latest baseline, there is a new failure on RTEMS:

```
[BEGIN] 04 TestChmod^M
OS_FileOpen_Impl():111:open(/RAM/Filename1): Invalid argument^M
[ FAIL] 04.001 file-api-test.c:257 - status after creat = -1^M
[ FAIL] 04.002 file-api-test.c:259 - status after close = -16^M
```

**To Reproduce**
Build with unit tests and execute on RTEMS 4.11 + pc686 + QEMU as documented in README.  The chmod test fails.

**Expected behavior**
Test should pass.

**System observed on:**
RTEMS 4.11.3, pc686 BSP, running in QEMU

**Additional context**
Appears to be introduced in PR #661 

Reinforces need (again) for CI script that runs the RTEMS build.

**Reporter Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2020-12-11:
I took a quick look, I think the issue is related to the way the test file is initally created:

https://github.com/nasa/osal/blob/9407cdf0be77521d850c587c0f749779c4656c3e/src/tests/file-api-test/file-api-test.c#L256

It doesn't make sense to use `OS_FILE_FLAG_CREATE | OS_FILE_FLAG_TRUNCATE` in combination with `OS_READ_ONLY` ... I believe the OS is probably catching this as invalid.

zanzaben, 2020-12-14:
Also dug into it and the FileOpen_Impl has a comment saying the same thing about read only. I can change the test to start with Read Write instead if you think that will fix the problem.

skliper, 2021-01-04:
What's the status here?  Can we get a fix in this round?

PR #749 (2021-01-25): Fix #685 change file create to use read write
**Describe the contribution**
fixes #685
Changed the file create to read write to work on RTEMS

**Testing performed**
Ran the test on RTEMS 5

**Expected behavior changes**
Test should pass

**System(s) tested on**
Ubunto 20.04, RTEMS 5

**Contributor Info - All information REQUIRED for consideration of pull request**
Alex Campell GSFC
commit 0ba7e879 (2021-01-25): Merge pull request #749 from zanzaben/fix685_RTEMS_chmod_test

Fix #685 change file create to use read write
commit 9074cf98 (2021-01-25): Fix #685, change file create to use read write
PR #767 (2021-01-27): osal Integration candidate: 2021-01-19
**Describe the contribution**

Fix #746, Add UtDebug message from OS_printf stub
Fix #692, Resolve typos in unit test assert OR statements
Fix #577, Document nested tests not supported
Fix #753, Remove UT_CheckForOpenSockets
Fix #229, add mqueue functional test
Fix #757, Readd extern C in osapi.h to support C++ use
Fix #692, display message when not implemented error occurs
Fix #591, add test teardown failure to test summary
Fix #699, format printf correctly
Fix #685, change file create to use read write
Fix #748, add additional casts
Fix #743, Create Sec [... cut]
closed 2021-01-27 by astrogeco