nasa/osal #705: Tests that use network stack do not run on RTEMS

jphickey, 2020-12-22:
**Is your feature request related to a problem? Please describe.**
The RTEMS network stack needs to be explicitly initialized in a platform-specific manner.  However, the barebone/minimal BSP that runs the unit tests does not do this initialization.  Therefore when running the network tests, it just gives an immediate error:

```
[BEGIN] 01 TestDatagramNetworkApi
rtems-net: network sema obtain: network not initialised

fatal error, exiting
```

**Describe the solution you'd like**
Not clear at this point.  The network init is quite board specific (refers to a specific nic driver) so probably doesn't fit as part of the OSAL BSP.  Suggest just documenting this limitation.

**Additional context**
As the network tests only use the loopback address (127.0.0.1) it might be possible to bring up the RTEMS network stack with only this interface and no "real" nic.  That might be a possibility to get these tests to run without making the BSP too complicated.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

skliper, 2020-12-28:
I like the suggestion of just bringing up the loopback interface for the OSAL BSP.

jphickey, 2020-12-28:
> I like the suggestion of just bringing up the loopback interface for the OSAL BSP.

Yes, this would be a good solution if it can work.  But we need to:

1. determine if its possible at all (i.e. does RTEMS have a null device/loopback only mode for its IP stack)
2. figure out how to make it work/coexist/not interfere in cases where a real BSP is used that has a real NIC.