nasa/osal #707: Race condition in "select" test

jphickey, 2020-12-28:
**Describe the bug**
Running the OSAL select test, I ran into a deadlock situation where the "multi" test got stuck and never finished.

**To Reproduce**
Hit or miss... Run test repeatedly on a system with other loads (e.g. parallel builds)

**Expected behavior**
Test should complete

**Code snips**
Checking the test status/backtrace it looks like two tasks (main + "Server_Fn") are waiting on the binary sem.  In particular the Server_Fn is stuck here:

https://github.com/nasa/osal/blob/d698a4d7ddc66826a4c7c287468b5f1aa6ca372f/src/tests/select-test/select-test.c#L162

While the main task is waiting in the teardown code (the `TestSelectMultipleRead` has completed, and it has invoked `Teardown_Multi` which in turn invokes `Teardown_Single` here):

https://github.com/nasa/osal/blob/d698a4d7ddc66826a4c7c287468b5f1aa6ca372f/src/tests/select-test/select-test.c#L273

**System observed on:**
Ubuntu 20.04

**Additional context**
This is likely related to the use of `OS_BinSemFlush`.  We should probably deprecate this function, as I cannot see how this can ever be used safely without it being a race condition.  VxWorks offers it which (I think) is why OSAL also offers it, but its a fundamentally broken concept.

I can confirm that looking at the traceback in gdb, the flush_count is indeed already 1 - meaning the flush had already happened by the time the Server_Fn entered the bin sem take routine.

**Reporter Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2020-12-28:
Discovered another race condition in here - this uses the same TCP port number as the "network-api-test" does.  This means when running ctest with a "-j8" option (parallel) that it might run at the same time as network-api-test, which causes it to fail.

Still not clear on why this wasn't _part of_ the network-api-test as I originally suggested - which would have avoided this - but nonetheless it has been implemented separately so this becomes a race condition issue.  Should use a different port number for this test so it doesn't interfere with the other test.

zanzaben, 2020-12-29:
It was not put into the network api test because not all the select tests use network features. As for selecting a new port number, is there a list of all the ports that are already being used or do I just have to search through the tests to find them.

PR #721 (2021-01-06): Fix #707 change sem flush to solve race condition
**Describe the contribution**
fixes #707
Removed Sem flush and changed it to sem give to prevent a rare race condition. 
Change the port numbers to be different from network test for when test are run in parallel. 

**Testing performed**
Ran the unit tests in parallel 

**Expected behavior changes**
Test shouldn't lock or fail 

**System(s) tested on**
Ubuntu 20.04

**Contributor Info - All information REQUIRED for consideration of pull request**
Alex Campbell GSFC
commit 13461bde (2021-01-06): Merge pull request #721 from zanzaben/fix707_Select_Test_Race_condition

Fix #707 change sem flush to solve race condition
commit ae888d44 (2021-01-06): Fix #707 Resolve issues of auto tests in parallel build
change sem flush to solve race condition
change port numbers to be different from network test
PR #744 (2021-01-07): osal Integration Candidate: 2021-01-05
**Describe the contribution**

Fix #574, Document UtAssert_Message parameters
Fix #598, Correct UtAssert_ZERO description typo`
Fix #707, Resolve issues of auto tests in parallel build
Fix #726, Resolve user's guide generation warnings

**Testing performed**
See Github Actions <https://github.com/nasa/cFS/pull/172/checks>

**Expected behavior changes**

PR #713 - Document UtAssert_Message parameters, also adds "see also" note for helper macros.

PR #714 - Fix doxygen typo

PR #721 - Replace `OS_BinSemFlush` with `OS_BinSemGive` to prevent a rare race cond [... cut]
closed 2021-01-07 by astrogeco