nasa/osal #718: Deprecate OS_fsBlocksFree and OS_fsBytesFree

jphickey, 2020-12-29:
**Is your feature request related to a problem? Please describe.**
As noted in #573 these functions return the value directly, which does double duty as an error code and therefore is limited to being `int32` - but this type is not big enough for large file systems.

**Describe the solution you'd like**
Deprecate these functions in favor of `OS_FileSysStatVolume()` added in PR #717, which returns the information as a OS_statvfs_t object containing all relevant information.

**Additional context**
References to these functions within OSAL tests need to be fixed/replaced, as well as references in CFE/apps.  This is a follow-on/completion of the fix for #573 / PR #717.  Deprecation has to be done separately because the new function needs to exist before the refs can be changed.

**Requester Info**
Joseph Hickey, Vantage Systems. Inc.

skliper, 2021-01-11:
@astrogeco @jphickey any way we can get a PR in this week? Delaying code review a week for this isn't very palatable, and I'd like to tag/branch and then remove all deprecated code ASAP.

skliper, 2021-01-11:
@ejtimmon - one last (so far) impact to apps in OSAL, see cfs_fm/fsw/src/fm_cmds.c

PR #759 (2021-01-21): Fix #718, deprecate OS_fsBlocksFree and OS_fsBytesFree
**Describe the contribution**
Deprecates the `OS_fsBlocksFree()` and `OS_fsBytesFree()` functions because the return value is inadequate for larger file systems.

Users should call OS_FileSysStatVolume() and read the respective output members of the structure as indicated.

Fixes #718 

**Testing performed**
Build CFE and unit test with both OMIT_DEPRECATED as true and false

**Expected behavior changes**
Any calls to these functions will become an error if building with `OMIT_DEPRECATED=true`

**System(s) tested on**
Ubuntu 20.04

**Contr [... cut]
commit 8c7535c5 (2021-01-21): Fix #718, deprecate OS_fsBlocksFree and OS_fsBytesFree

Users should call OS_FileSysStatVolume() and read the respective
output members of the structure as indicated.
commit ff42bfd6 (2021-01-21): Merge pull request #759 from jphickey/fix-718-deprecate-bytesblocksfree

Fix #718, deprecate OS_fsBlocksFree and OS_fsBytesFree
PR #767 (2021-01-27): osal Integration candidate: 2021-01-19
**Describe the contribution**

Fix #746, Add UtDebug message from OS_printf stub
Fix #692, Resolve typos in unit test assert OR statements
Fix #577, Document nested tests not supported
Fix #753, Remove UT_CheckForOpenSockets
Fix #229, add mqueue functional test
Fix #757, Readd extern C in osapi.h to support C++ use
Fix #692, display message when not implemented error occurs
Fix #591, add test teardown failure to test summary
Fix #699, format printf correctly
Fix #685, change file create to use read write
Fix #748, add additional casts
Fix #743, Create Sec [... cut]
closed 2021-01-27 by astrogeco