nasa/osal #765: API's Null pointer check in void methods

zanzaben, 2021-01-20:
**Is your feature request related to a problem? Please describe.**
Some pointers are used in methods that return void so the OS_CHECK_POINTER function doesn't work in those methods since it returns an error code. There needs to be a way to check that pointers are valid inside void methods. 

**Describe the solution you'd like**
Need to discuss the best solution 

**Additional context**
related to #742, That branch has comments marking everywhere this comes up. 

**Requester Info**
Alex Campbell GSFC

jphickey, 2021-01-20:
You should be able to use `BUGCHECK()` macro directly but leave the second argument blank, i.e.:

`BUGCHECK(ptrarg != NULL,)`

Note the comma because its a two-arg macro, but you don't actually have to use the second arg, but it has to be there syntactically.

However in general `void` functions are less common - and it calls for the question:

1. should it be considered an error in the first place?  (maybe it should just accept all inputs)
2. otherwise, then should it really be returning an `int32` so it can communicate to the caller that the function didn't do its job?

By design if a function returns `void` it is basically saying _there is no way this function can fail_.  It's good and simple and I like these type of functions - but if the implementation actually _can_ fail but cannot communicate that failure - then that seems more like an API design flaw.  A void function should always accept its inputs and do its performed task (item 1 above) no matter what.  If it doesn't meet that model then it should probably be returning an error code (item 2 above).

skliper, 2021-01-21:
Interesting topic... let's go with the BUGCHECK for now.  We can trade changing function returns for APIs that can only fail when a bad pointer is passed in at a later date.

jphickey, 2021-01-21:
FWIW I think most (if not all?) of the functions that fall into this category are internal/helper functions and therefore should just do what they are told (case 1).  If it were a public API then it should probably return something (case 2).    

At this point its not clear to me if there are any public APIs that are affected - so there may not be any real issue here...

zanzaben, 2021-01-21:
After going through and removing the checks from internal methods there are still 2 cases. 

https://github.com/nasa/osal/blob/bfca5b21f5fc2032505256fc67ae974f53a2a319/src/os/shared/src/osapi-idmap.c#L1428-L1454

https://github.com/nasa/osal/blob/bfca5b21f5fc2032505256fc67ae974f53a2a319/src/os/shared/src/osapi-printf.c#L245-L306

jphickey, 2021-01-21:
Yeah for those two it should be possible to use the BUGCHECK macro but leave the second argument blank.  That way it basically turns into a simple assert (depending on whether strict checking is enabled in the config).

PR #766 (2021-02-03): Fix #742, make sure all pointers are checked for null
**Describe the contribution**
Fixes #742 
Went through all the api's and made sure all pointers have a null check, or a comment stating that it can be null. 

Fixes #765 

**Testing performed**
Build and run unit test

**Expected behavior changes**
none

**System(s) tested on**
Ubuntu 20.04

**Additional context**
13 To Do left in here that will be fixed by #765 

**Contributor Info - All information REQUIRED for consideration of pull request**
Alex Campbell GSFC
commit 3b2c7298 (2021-02-03): Fix #765, add null pointer check.
PR #774 (2021-02-12): osal Integration candidate: 2021-02-02
**Describe the contribution**
Fix #742, make sure all pointers are checked for null
Fix #742, remove null checks from internal methods.
Fix #602, bring OSAL code coverage back up to 100%
Fix #765, add null pointer check.
Fix #771, Add workflow timeout and format check 
Fix #777, Static allocation of DummyVec in OSC_INUM_TO_IVEC stub
Fix #783, Initialize locals flagged in static analysis
Fix #785, Clean commented out code
Fix #788, Resolve coercion alters value warnings
Fix #790, Return status from OS_ConsoleAPI_Init
Fix #797, Update os-impl-no-symtab.c to [... cut]
closed 2021-02-12 by astrogeco