nasa/osal #777: OSC_INUM_TO_IVEC stub returning stack allocated memory

skliper, 2021-02-09:
**Is your feature request related to a problem? Please describe.**
CodeQL warning on line 72 (returning VecTbl):
https://github.com/nasa/osal/blob/09a2c5e793639e508174a2446989274acf36f25e/src/unit-test-coverage/ut-stubs/src/vxworks-intLib-stubs.c#L51-L77

**Describe the solution you'd like**
Make DummyVec static

**Describe alternatives you've considered**
Remove, no longer used in OSAL testing but could be used by PSP implementations.  Leaving in for now.

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

astrogeco, 2021-02-12:
>Describe alternatives you've considered
Remove, no longer used in OSAL testing but could be used by PSP implementations. Leaving in for now.

@skliper should we open a separate issue to remove this or port it to PSP?

skliper, 2021-02-12:
@astrogeco Probably not at this point?  I'm fine w/o for now.  If/when someone needs in a PSP it'll be here/available.

astrogeco, 2021-02-12:
I'd like having the issue to remove it so we don't forget

jphickey, 2021-02-12:
> I'd like having the issue to remove it so we don't forget

I think the consensus is _not_ to remove it for now, as it could prove useful again at some point, such as for a VxWorks PSP that does use interrupts.

PR #778 (2021-02-12): Fix #777, Static allocation of DummyVec in OSC_INUM_TO_IVEC stub
**Describe the contribution**
Fix #777, declares DummyVec as static to avoid warning (and returning stack allocated memory) when returning VecTbl.

**Testing performed**
Build and execute unit tests, passed.

**Expected behavior changes**
Bug in stub fixed (could have been issues if return value used).

**System(s) tested on**
 - Hardware: cFS Dev Server
 - OS: Ubuntu 18.04
 - Versions: cFS Bundle + this commit

**Additional context**
CodeQL warning resolution

**Third party code**
None

**Contributor Info - All information REQUI [... cut]
commit 68d0e7f5 (2021-02-12): Merge pull request #778 from skliper/fix777-inum_to_ivec

Fix #777, Static allocation of DummyVec in OSC_INUM_TO_IVEC stub
commit a51fc93b (2021-02-12): Fix #777, Static allocation of DummyVec in OSC_INUM_TO_IVEC stub
PR #774 (2021-02-12): osal Integration candidate: 2021-02-02
**Describe the contribution**
Fix #742, make sure all pointers are checked for null
Fix #742, remove null checks from internal methods.
Fix #602, bring OSAL code coverage back up to 100%
Fix #765, add null pointer check.
Fix #771, Add workflow timeout and format check 
Fix #777, Static allocation of DummyVec in OSC_INUM_TO_IVEC stub
Fix #783, Initialize locals flagged in static analysis
Fix #785, Clean commented out code
Fix #788, Resolve coercion alters value warnings
Fix #790, Return status from OS_ConsoleAPI_Init
Fix #797, Update os-impl-no-symtab.c to [... cut]
closed 2021-02-12 by astrogeco