nasa/osal #808: Convert from strlen to strnlen where appropriate

skliper, 2021-02-11:
**Is your feature request related to a problem? Please describe.**
Per security standards strlen should be avoided if possible:
```
src/os/vxworks/src/os-impl-shell.c:        OS_write(fdCmd, Cmd, strlen(Cmd));
src/os/vxworks/src/os-impl-symtab.c:    if (strlen(name) >= OS_MAX_SYM_LEN)
src/os/shared/src/osapi-sockets.c:        len = strlen(sock->stream_name);
src/os/shared/src/osapi-sockets.c:        len = strlen(sock->stream_name);
src/os/shared/src/osapi-idmap.c:            name_len = strlen(record->name_entry);
src/os/shared/src/osapi-filesys.c:    mplen = strlen(filesys->virtual_mountpt);
src/os/shared/src/osapi-filesys.c:    if (strlen(dev_name) >= OS_FS_DEV_NAME_LEN)
src/os/shared/src/osapi-filesys.c:    VirtPathLen = strlen(VirtualPath);
src/os/shared/src/osapi-filesys.c:    if (strlen(name_ptr) >= OS_MAX_FILE_NAME)
src/os/shared/src/osapi-filesys.c:            SysMountPointLen = strlen(filesys->system_mountpt);
src/os/shared/src/osapi-filesys.c:            VirtPathBegin    = strlen(filesys->virtual_mountpt);
```

**Describe the solution you'd like**
Replace with strnlen where possible.

**Describe alternatives you've considered**
None

**Additional context**
Security analysis warning

**Requester Info**
Jacob Hageman - NASA/GSFC

jphickey, 2021-02-11:
Unfortunately `strnlen()` is not C99... otherwise would have done this a long time ago!

My recommendation is to use `memchr()` which returns a pointer to the null byte.  If this returns non-null the string pointer can be subtracted which gets the length.  It requires an extra local variable but memchr() is part of ISO C standard so every platform should have this.

skliper, 2021-02-11:
That's what I get for copy pasting from a security analysis report, suggestion noted!

jphickey, 2021-02-11:
We could always make an OSAL wrapper around `memchr()` to give ourselves a utility function that works like strnlen does.  That would be easy.  Maybe in `os-common.c` ....   then even other stuff like CFE could use it too.

I can easily see that turning into a dumping ground for "functions the ISO C standard should have but doesn't"

PR #810 (2021-02-12): Fix #808, length-limited string length checks
**Describe the contribution**

Create a wrapper around `memchr()` that mimics the non-C99 function `strnlen()` which is defined in POSIX-2008.

Use this instead of `strlen()` whenever the string being checked either originates in or will be copied into a fixed-length array buffer.

Fixes #808 

**Testing performed**
Build and sanity check CFE
Run all unit tests on both native and RTEMS

**Expected behavior changes**
No behavior changes except if a bug causes strings to be unterminated.

**System(s) tested on**
Ubuntu 20.04 (native)
RTEMS 4.11.3 (qe [... cut]
commit 842fe880 (2021-02-12): Merge pull request #810 from jphickey/fix-808-strnlen

Fix #808, length-limited string length checks
commit efa18440 (2021-02-12): Fix #808, length-limited string length checks

Create a wrapper around memchr() that mimics the non-C99 function
"strnlen()" which is in POSIX-2008.

Use this instead of strlen() whenever the string being checked
either originates in or will be copied into a fixed-length array buffer.
PR #774 (2021-02-12): osal Integration candidate: 2021-02-02
**Describe the contribution**
Fix #742, make sure all pointers are checked for null
Fix #742, remove null checks from internal methods.
Fix #602, bring OSAL code coverage back up to 100%
Fix #765, add null pointer check.
Fix #771, Add workflow timeout and format check 
Fix #777, Static allocation of DummyVec in OSC_INUM_TO_IVEC stub
Fix #783, Initialize locals flagged in static analysis
Fix #785, Clean commented out code
Fix #788, Resolve coercion alters value warnings
Fix #790, Return status from OS_ConsoleAPI_Init
Fix #797, Update os-impl-no-symtab.c to [... cut]
closed 2021-02-12 by astrogeco