nasa/osal #821: OSAL should have a "OS_GetVersionString()" function

jphickey, 2021-02-17:
**Is your feature request related to a problem? Please describe.**
Currently the OSAL version is a string that is made available as a macro - `OS_VERSION_STRING` - which defined in `osapi-version.h`

The problem with this is that because its a preprocessor macro - the replacement is done at the point the macro is _referenced_ and therefore the value ends up getting built into the binary code for whatever referenced this (in CFE use case this is `cfe_es_task.o`)

This means if that code is later linked with a different/updated OSAL library (but the CFE code using OSAL is **not** recompiled) it will still report the original OSAL version string, which is wrong.

**Describe the solution you'd like**
There should be a function call into OSAL to get the version string, rather than a macro in a header.  Therefore if/when re-linking without recompilation, it will report the correct value.

**Additional context**
In summary - the OSAL version string should be stored within the OSAL library binary file, rather than stored within the CFE library binary file.  This is confirmed via the "strings" tool that right now its in the wrong place:

    $ strings build/native/default_cpu1/cfe-core/libcfe-core.a | grep osal
    cFS Versions: cfe v6.8.0-rc1+dev348, osal v5.1.0-rc1+dev262, psp

This shows that the actual string showing the OSAL version as reported by the cFE event is baked into the CFE library, not the OSAL library where it should be.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2021-02-17:
Suggestion is to add a new OSAL API, something simple like:

    const char *OS_GetVersionString(void);

Which just returns the value of the `OS_VERSION_STRING` macro.

skliper, 2021-02-17:
I vote we fix for Caelum.

jphickey, 2021-02-17:
> I vote we fix for Caelum.

OK!  It's trivial enough, I can do this real quick.

commit 6197a52f (2021-03-02): Fix #821, add accessor functions for version strings

Adds 4 version API calls:

const char *OS_GetVersionString(void);
const char *OS_GetVersionCodeName(void);
void OS_GetVersionNumber(uint8[4]);
uint32 OS_GetBuildNumber(void);

These return the values of current macros in osapi-version.h.

The accessor function should be the preferred way to get the OSAL version
info (vs. using macro directly) as it is evaluated at OSAL library
compile time, rather than application compile time, and thus will
remain correct in the event that OSAL is relinked without recompiling
the application.
commit 7c47365b (2021-03-02): Fix #821, add ut stubs for version API
commit 86b220b8 (2021-03-02): Merge pull request #824 from 'jphickey/fix-821-version-strings'

Fix #821, add accessor functions for version strings

Fix format in `osapi-utstub-version.c`
PR #824 (2021-03-02): Fix #821, add accessor functions for version strings
**Describe the contribution**
Adds 3 simple API calls:

    const char *OS_GetVersion(void);
    const char *OS_GetVersionCodeName(void);
    const char *OS_GetVersionDescription(void);

These directly return the values of string macros defined in `osapi-version.h`.

The accessor function should be the preferred way to get the OSAL version string (vs. using macro directly) as it is evaluated at OSAL library compile time, rather than application compile time, and thus will remain correct in the event that OSAL is relinked without recompiling  [... cut]
closed 2021-03-04 by astrogeco