nasa/osal #866: Scrub use of () in return statements

skliper, 2021-03-10:
**Is your feature request related to a problem? Please describe.**
Inconsistent use of (), style.  Example below but should fix all.
https://github.com/nasa/osal/blob/ead5723f59ee49b36a36d6cb66c4cfe680ebbd68/src/os/portable/os-impl-no-symtab.c#L50-L60

**Describe the solution you'd like**
Fix.

**Describe alternatives you've considered**
Leave as-is (future work)

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC, OSAL code review

thnkslprpt, 2022-09-26:
Is there consensus on removing the parentheses? @astrogeco @skliper @jphickey 

There is ~400 instances of unnecessary parentheses surrounding the return values in the cFS codebase.
For comparison, there is ~4,000 instances of return statements that do not use parentheses for the return value.

From what I can tell no modern C/C++ coding standards require, or even prefer, parentheses around return values (e.g. MISRA, JSF, JPL, ESA etc.).

I'm happy to work on this but wanted to confirm if it's agreed upon first :-)

skliper, 2022-09-26:
@dmknutsen - your call.  Simple style/consistency thing but could impact many repos.  Looks like a simple sed one-liner or similar to fix?

thnkslprpt, 2022-09-26:
OK Thanks Jake/Dan.
I will take care of it.

thnkslprpt, 2022-09-27:
For now, I will update only the return statements that return single values/terms like the one Jake linked to above.

I will leave the complex statements as is, even though they are logically unnecessary as well. For example:
https://github.com/nasa/osal/blob/38559d447fdeb1a85f8bb09938df30d7a50918a4/src/os/shared/src/osapi-time.c#L206-L210

I know some people prefer parentheses in these cases to make the arithmetical logic easier to break down in their head.

Happy to update the rest in future if it's considered prudent.

commit b3c3cc25 (2022-10-03): Merge pull request #1292 from thnkslprpt/fix-866-return-parentheses

Fix #866, Remove unnecessary parentheses around return values.
commit dcb1d6be (2022-10-03): Fix #866, Remove unnecessary parentheses around return values.
PR #1292 (2022-10-03): Fix #866, Remove unnecessary parentheses around return values.
**Checklist (Please check before submitting)**
* [x] I reviewed the [Contributing Guide](https://github.com/nasa/osal/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
Fixes #866
Removes parentheses in return statements in OSAL that return a single value/term.
This is aligns these retur [... cut]
closed 2022-10-03 by dzbaker