nasa/osal #880: Fix constraints/bounds in OS_ObjectIdFindNextFree

skliper, 2021-03-11:
**Is your feature request related to a problem? Please describe.**
The local_id is constrained by serial, which is constrained by max_id. Then serial is further constrained by local_id. Seems circular:
https://github.com/nasa/osal/blob/ead5723f59ee49b36a36d6cb66c4cfe680ebbd68/src/os/shared/src/osapi-idmap.c#L661

Bounds check?
https://github.com/nasa/osal/blob/ead5723f59ee49b36a36d6cb66c4cfe680ebbd68/src/os/shared/src/osapi-idmap.c#L666

**Describe the solution you'd like**
Scrub logic and fix if needed, and/or add comments if they would help.  If not in error remove bug label.

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC, OSAL code review

jphickey, 2021-03-12:
Looked into the logic again, seems OK to me.  The limits/values refer to different things:

- `serial` refers to the last "object serial number" which is in the range of [0 .. 0xFFFF].  Each allocation starts at the serial number of the last allocated ID - not zero.
- `max_id` refers to the total number of _real_ entries in the table.  If a free slot is not found within this number of checks, there is no free slot.
- `local_id` refers to the current position in the real table in the range of [ 0 .. max_id ].
- `base_id` refers to the base point in the global table for this type of object.  All resources in OSAL are stored in a large global object table.  Each type of object has a different base within that large table, forming a bunch of subtables, based on the maximum number of objects of that type.
- The lookup `&OS_common_table[local_id + base_id]` is for the large table and can never exceed the bounds of that table because `local_id` is always < `max_id` (which is the number of entries in the common table devoted to this object type).

So it all looks OK to me, if any real issue/unclear items remain, please comment.

klystron78, 2021-03-23:
if local_id is < max_id, what prevents (local_id + base_id) from being > max_id?

jphickey, 2021-03-23:
> if local_id is < max_id, what prevents (local_id + base_id) from being > max_id?

It likely will be - but that's not a problem....

`OS_common_table` is a big/unified table - each resource type is given a "slice" of that table for its records.  The slice starts at `base_id` and is `max_id` entries long.

`local_id` refers to the offset within the slice i.e. relative to the `base_id`.  It is computed by modulus with `max_id` (so that it is always < `max_id`).  The index `base_id + local_id` will always be within the bounds of the large `OS_common_table`.

klystron78, 2021-03-23:
I guess I thought there was a range of IDs in the table for each object type. It's ok if it goes out of that range as long as it stays inside the table?

jphickey, 2021-03-24:
> I guess I thought there was a range of IDs in the table for each object type. It's ok if it goes out of that range as long as it stays inside the table?

I'm not following ... yes there is a range for each object type.   But it never goes "out of range".

There is the size of the overall table (OS_common_table) and the size of the _slice_ of that table that is devoted to the current type of object.  Index `base_id` is the starting position of the slice in the large table, `max_id` is the size of the slice, and therefore `base_id + max_id` is one past the end of the slice.  Index `base_id + local_id` refers to an entry of the current slice, and will never go beyond/out of bounds because `local_id` is always less than `max_id`.

klystron78, 2021-03-24:
If you're sure it'll never go out of its range then I won't bother analyzing it. I was asking questions to decide if I should isolate the code and test it just to be helpful. I'm having a hard time understanding what serial is used for the way it is. It seems to me like that wouldn't be necessary and we should just keep an index for each object type. I'm not proposing any change though. I was asked to take another look at it and I did. :)

skliper, 2021-03-25:
I just worked through the math and it looks OK to me.  The way I understand it is serial expands the id space to avoid id reuse (for up to 0xFFFF elements) where use of the index would reuse id's as soon as you wrap on max_ids (the slice size allocated to that type).  max_ids can be small...  Just provides a bit more resiliency.

jphickey, 2021-03-25:
Yes, confirmed -- the point is to not recycle IDs.  For instance if there were 10 slots in the array/table for a given object type, an allocation might find index "4" is available and use that slot.  If that object is later deleted, slot 4 becomes open again.  The next time something is allocated again, it will find slot "4" open and use it again.

With the serial number concept - that new object gets assigned 14, not 4, but still maps to the same slot.  So an old/stale reference to "4" will be seen as invalid (ID 4 was deleted).  

Without the serial number, the new ID was assigned "4" again and the new object was an alias to the old one.  Bad.

With serial numbers the objects don't repeat until 0xFFFF - so you have to create and delete thousands of objects before you'll see a repeat ID.

skliper, 2021-03-25:
Closing - works as-is.

closed 2021-03-25 by skliper