nasa/osal #881: Avoid collisions in key value generation

skliper, 2021-03-11:
**Is your feature request related to a problem? Please describe.**
Could be collisions in key value:
https://github.com/nasa/osal/blob/ead5723f59ee49b36a36d6cb66c4cfe680ebbd68/src/os/shared/src/osapi-idmap.c#L742-L749

**Describe the solution you'd like**
Analyze and fix if needed.  Add comments if it would help, remove bug label if not a real issue.

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC, OSAL code reivew

jphickey, 2021-03-12:
Not a bug - this key value is only to create a value which is unlikely to occur at random.  It is used as a sanity check between locks and unlocks.  There is no issue/problem if there is a collision, except if they occur too often it defeats the purpose (but this won't collide often)

skliper, 2021-03-16:
@jphickey @klystron78 - Can this be close or is there something remaining to be fixed?

jphickey, 2021-03-16:
I think we should close, I don't see an issue here.

klystron78, 2021-03-23:
Why not replaced the fixed value in the MSB with an 8-bit rollover counter? We could even keep the counter in the table object so that once lock is achieved, it can increment and be saved for context.

jphickey, 2021-03-23:
The only point of the fixed value MSB is to make sure some bits are always set, i.e. such that the key value is never totally zero - because zero would alias to `OS_LOCK_KEY_INVALID`.  If the MSB was replaced by anything that varies, there is a chance it could be zero.

Again, the objective of this key is just to sanity check that every task/operation that initiates a transaction in OSAL global tables is also properly finishing (or canceling) that transaction, and that there are no rogue ops doing things improperly.  This key is just a value that should be consistent between the start and end to facilitate that sanity check, or serve as evidence/breadcrumbs in a debugger if something did not follow correct procedure.  24 bits of variation in that key (16.7M values) is plenty to ensure this and meet that objective.  This isn't a crypto-level check, its just for sanity/catching mistakes/etc.

The fixed upper 8 bits serve as useful identifying info, in addition to ensuring the key is nonzero.  For instance if someone overwrote the OS_object_token_t value and clobbered this key, the clobbered value probably wouldn't start with `OS_LOCK_KEY_FIXED_VALUE` anymore, and so it would be more obvious that it was clobbered.

klystron78, 2021-03-23:
Any of the 3 other bytes could be overwritten as well. Perhaps via a misplaced single-byte write. I understand the point of this, though. If a collision occurs you only risk maybe not printing a debug message should the non-owner try to unlock. You would hope to find that case long before a collision might occur.

Another way to possibly add robustness without any real cost would be a second value that's perhaps 2s complement of the original value. Then check that they match.

Well, I was asked to comment so I commented. :)

skliper, 2021-03-24:
I was probably too open ended in my request for comments, in that I'm not really looking for ways to make it "better" or fishing for improvement possibilities at this point.  More from the context of a code review for the certification activity and is it "good enough".  As in does the current implementation makes sense, will perform as advertised, etc.  We can always open issues for future work to address additional robustness, trade added complexity against added value, etc but those don't need to be implemented/resolved for cert as long as the current implementation is sufficient.

klystron78, 2021-03-24:
I mean sure it works. Of course you could just skip all of it and depend on the implementation of the mutex to only allow release from the owning thread. Of course this, as we saw, varies on POSIX.

skliper, 2021-03-24:
Closing in the context of Caelum code review (works as intended).  Feel free to open a follow-on issue to discuss alternative implementations if desired.

closed 2021-03-24 by skliper