nasa/osal #888: Report symEach errors in OS_SymbolTableDump_Impl

skliper, 2021-03-11:
**Is your feature request related to a problem? Please describe.**
Errors not reported:
https://github.com/nasa/osal/blob/ead5723f59ee49b36a36d6cb66c4cfe680ebbd68/src/os/vxworks/src/os-impl-symtab.c#L262

**Describe the solution you'd like**
Check for non-NULL, report error (write to file?), return error.

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC, OSAL code review

jphickey, 2021-03-17:
Looked up the VxWorks docs for `symEach` .... This function is defined a little oddly.  It returns NULL on _success_ and on an error it returns a pointer to the symbol it last visited.  However as far as I can tell the only way for this to happen is for the user callback function to return `false`.  

With the way the callback is currently implemented, all errors are currently reported at least via OS_DEBUG() and also possibly translated to an OS_ERROR response already.

It looks like this function can also return NULL if the symbol table ID itself was invalid - but this is getting the value direct from vxWorks (`sysSymTbl` global) and therefore cannot be bad.  It's also not clear now this is differentiated from the success case, since both return NULL.

My recommendation is that we leave the call to `symEach()` as it is.  The VxWorks API here is not ideal - return value is ambiguous (NULL can mean success or error, and also non-NULL does not necessarily mean an error).  It is being handled as well as can be done with the limitations of this underlying API design.

The only question I can see is whether exceeding the `size_limit` should be treated as an error or not. Currently it just stops at the size limit - there is a debug statement to say the limit was reached, but the overall function will still return `OS_SUCCESS`.  A case could be made that this should be an error.

skliper, 2021-03-17:
Ping @klystron78 - is this response sufficient to close the issue?

jphickey, 2021-03-17:
After reviewing the function here -- there are some better return codes we could propagate up the stack.  In particular I think if the size limit was insufficient we should not be returning OS_SUCCESS.  

I will submit a PR to address that shortly.

PR #915 (2021-03-29): Fix #888, better return codes from OS_SymbolTableDump_Impl
**Describe the contribution**
Improve the error codes from this function.

- Introduces a new error `OS_ERR_OUTPUT_TOO_LARGE` if the size limit was insufficient (instead of `OS_SUCCESS`).
- Return `OS_ERROR` if an empty file was written - this likely indicates some fundamental issue with the VxWorks symbol table.
- Return `OS_ERR_NAME_TOO_LONG` if one of the symbol names was too long, (instead of generic `OS_ERROR`).

Improve unit test to check for/verify these responses.

Fixes #888

**Testing performed**
Build and sanity check, run a [... cut]
commit 318afe6e (2021-03-29): Merge pull request #915 from jphickey/fix-888-symEach-errs

Fix #888, better return codes from OS_SymbolTableDump_Impl
commit c9327ee3 (2021-03-29): Fix #888, better return codes from OS_SymbolTableDump_Impl

Improve the error codes from this function.

- Introduces a new error "OS_ERR_OUTPUT_TOO_LARGE" if the size
limit was insufficient (instead of OS_SUCCESS).
- Return OS_ERROR if an empty file was written - this likely
indicates some fundamental issue with the VxWorks symbol table.
- Return OS_ERR_NAME_TOO_LONG if one of the symbol names was
too long, (instead of generic OS_ERROR).

Improve unit test to check for/verify these responses.
PR #927 (2021-04-01): osal Integration candidate: 2021-03-30
## Describe the contribution
Fix #843, remove BIG/LITTLE bit order macros 
Fix #846, Minor clean up and clarification in comments/naming
Fix #831, Finish os-impl-bsd-socket.c coverage testing 
Fix #888, better return codes from OS_SymbolTableDump_Impl
Fix #853, remove OS_TaskRegister
Fix #934, Remove unused SCRIPT_MODE flag
Fix #929, use test function for osal_id_t 

## Testing performed
osal checks <https://github.com/nasa/osal/pull/927/checks>
bundle checks <https://github.com/nasa/cFS/pull/227/checks>

## Expected behavior changes

### PR #878

Removes [... cut]
closed 2021-04-01 by astrogeco