nasa/osal #904: A bug in OS_TaskExit workwith vxworks

sfzyk, 2021-03-14:
very sorry to bother you
i work with vxwork6.7 
got an error when task exiting 
OS_Unlock_Global():730:ERROR: global 1 released by task 0xffffffff when owned by task <my_task_id>

when exiting a task , in OS_TaskExit
OS_ObjectIdFinalizeDelete(OS_SUCCESS, &token); will change final id to UNDEFINED

in OS_ObjectIdFinalizeDelete
OS_ObjectIdTransactionFinish change  task_record_active_id to final_id(i.e UNDEFINED)
and then OS_Unlock_Global(token->obj_type)

but when OS_lock_Global(token->obj_type) is called task id is true task id
OS_Unlock_Global(token->obj_type)  get taskid from task record active id (i.e UNDEFINED) So produce an error 

Maybe changing OS_ObjectIdFinalizeDelete(OS_SUCCESS, &token) to  OS_ObjectIdFinalizeDelete(OS_ERROR, &token) help?

very very appreciate your reply

jphickey, 2021-03-15:
Can you confirm which baseline/version of OSAL you are running?

This sounds like a duplicate of #645 which was fixed by PR #704 a couple months ago.  If your baseline is older than this please update to latest and re-check.

sfzyk, 2021-03-16:
Thanks! i'm running OSAL4.x and fixed in 5.0.0 by your pr :) ！

closed 2021-03-16 by sfzyk