nasa/osal #916: Comments on header guard endif don't match ifdef

skliper, 2021-03-17:
**Is your feature request related to a problem? Please describe.**
For example:
https://github.com/nasa/osal/blob/ead5723f59ee49b36a36d6cb66c4cfe680ebbd68/src/os/inc/common_types.h#L31-L32
https://github.com/nasa/osal/blob/ead5723f59ee49b36a36d6cb66c4cfe680ebbd68/src/os/inc/common_types.h#L191

**Describe the solution you'd like**
Comments should be consistent w/ code

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

prsndp, 2021-03-25:
Hi @skliper Can I work on this issue if it's open??

skliper, 2021-03-25:
@prsndp Thanks for the offer, but I think a script was just put together to automate this fix.

PR #931 (2021-04-08): Fix #916, scrub include guards
**Describe the contribution**
Apply script to scrub all header file block comments and include guards on OSAL.

Fixes #916

**Testing performed**
Build and sanity check, run all unit tests

**Expected behavior changes**
None, comment and ifdef consistency fixups only

**System(s) tested on**
Ubuntu 20.04

**Contributor Info - All information REQUIRED for consideration of pull request**
Joseph Hickey, Vantage Systems, Inc.
commit 4ce8dec2 (2021-04-08): Fix #916, scrub headers in unit-test-coverage

Scrub all include guards and file scope doxygen comments in
src/unit-test-coverage
commit 6a3f0b7a (2021-04-08): Fix #916, scrub headers in src/bsp and src/os

Ensure all headers have a consistent include guard and file block comment
This gets all FSW code within src/os and src/bsp.
commit a8765199 (2021-04-08): Merge pull request #931 from jphickey/fix-916-include-guards

Fix #916, scrub include guards
closed 2021-04-08 by astrogeco