nasa/osal #942: Timer-add-api-test failing intermittently

zanzaben, 2021-04-05:
**Describe the bug**
On my virtual machine running Ubuntu 20.04, The Timer-add-api-test fails roughly 50% of the times I run it. 

**To Reproduce**
Run make test 

**Expected behavior**
It should pass 100% of the time. 

**Code snips**
Here is the printout from the test 

[BEGIN] 01 TestTimerAddApi
[ PASS] 01.001 timer-add-api-test.c:84 - OS_TimeBaseCreate() (0) == OS_SUCCESS
[ PASS] 01.002 timer-add-api-test.c:88 - OS_TimeBaseSet() (0) == OS_SUCCESS
[ PASS] 01.003 timer-add-api-test.c:93 - Timer 0 Created RC=0 ID=90001
[ PASS] 01.004 timer-add-api-test.c:93 - Timer 1 Created RC=0 ID=90002
[ PASS] 01.005 timer-add-api-test.c:93 - Timer 2 Created RC=0 ID=90003
[ PASS] 01.006 timer-add-api-test.c:93 - Timer 3 Created RC=0 ID=90004
[ PASS] 01.007 timer-add-api-test.c:113 - Timer 0 programmed RC=0
[ PASS] 01.008 timer-add-api-test.c:113 - Timer 1 programmed RC=0
[ PASS] 01.009 timer-add-api-test.c:113 - Timer 2 programmed RC=0
[ PASS] 01.010 timer-add-api-test.c:113 - Timer 3 programmed RC=0
[ INFO] timer-add-api-test.c:119:Starting Delay loop.

[ PASS] 01.011 timer-add-api-test.c:140 - Timer 0 delete RC=0. Count total = 53
[ PASS] 01.012 timer-add-api-test.c:140 - Timer 1 delete RC=0. Count total = 61
[ PASS] 01.013 timer-add-api-test.c:140 - Timer 2 delete RC=0. Count total = 29
[ PASS] 01.014 timer-add-api-test.c:140 - Timer 3 delete RC=0. Count total = 37
[ PASS] 01.015 timer-add-api-test.c:157 - Expected ticks = 61
[ FAIL] 01.016 timer-add-api-test.c:163 - Timer 0 count >= 58
[ PASS] 01.017 timer-add-api-test.c:164 - Timer 0 count <= 64
[ PASS] 01.018 timer-add-api-test.c:157 - Expected ticks = 71
[ FAIL] 01.019 timer-add-api-test.c:163 - Timer 1 count >= 68
[ PASS] 01.020 timer-add-api-test.c:164 - Timer 1 count <= 74
[ PASS] 01.021 timer-add-api-test.c:157 - Expected ticks = 34
[ FAIL] 01.022 timer-add-api-test.c:163 - Timer 2 count >= 31
[ PASS] 01.023 timer-add-api-test.c:164 - Timer 2 count <= 37
[ PASS] 01.024 timer-add-api-test.c:157 - Expected ticks = 44
[ FAIL] 01.025 timer-add-api-test.c:163 - Timer 3 count >= 41
[ PASS] 01.026 timer-add-api-test.c:164 - Timer 3 count <= 47
[ PASS] 01.027 timer-add-api-test.c:170 - OS_TimerAdd() (0) == OS_SUCCESS
[ PASS] 01.028 timer-add-api-test.c:175 - OS_TimerAdd() (-2) == OS_INVALID_POINTER
[ PASS] 01.029 timer-add-api-test.c:179 - OS_TimerAdd() (-16) == OS_ERR_INVALID_ID
[ PASS] 01.030 timer-add-api-test.c:183 - OS_TimerAdd() (-2) == OS_INVALID_POINTER
[ PASS] 01.031 timer-add-api-test.c:187 - OS_TimerAdd() (-15) == OS_ERR_NAME_TAKEN
[ PASS] 01.032 timer-add-api-test.c:191 - OS_TimerAdd() (-2) == OS_INVALID_POINTER
[  END] 01 TestTimerAddApi      TOTAL::32    PASS::28    FAIL::4     MIR::0     TSF::0     TTF::0     N/A::0   

COMPLETE: 1 tests Segment(s) executed

**System observed on:**
Ubuntu 20.04

**Reporter Info**
Alex Campbell GSFC

jphickey, 2021-04-05:
It was noted that this configuration involves a virtual machine, correct?

Can you share the details of your VM setup?  That is, what hypervisor software/version and what host OS/machine/physical CPU type?  I locally have both VirtualBox and QEMU available so I'd like to replicate your config as closely as I can to see if I can replicate your error.

FWIW, I regularly run the tests on various VMs running under QEMU/libvirt and have not seen failure as described, so I'm curious to investigate this.

zanzaben, 2021-04-06:
[image]
Here are my specs on my VM, if you want to know anything else just let me know.

skliper, 2021-08-12:
See comments on #943, the timing performance on the failing system is really poor (>500 ms errors vs the expected 1Hz cycle)