nasa/osal #943: OSAL Timer UT failing intermittently

zanzaben, 2021-04-05:
**Describe the bug**
On my virtual machine running Ubuntu 20.04, OSAL_timer_UT fails roughly 50% of the times I run it. Specifically the OS_TimerSet test. 

**To Reproduce**
Run Make test

**Expected behavior**
It should pass 100% of the time. 

**Code snips**
Here is the printout 

BEGIN] 05 OS_TimerSet
[ PASS] 05.001 ut_ostimer_timerio_test.c:525 - #1 Invalid-id-arg
[ INFO] 05.002 ut_ostimer_timerio_test.c:532 - #2 Internal-error
[ INFO] ut_ostimer_timerio_test.c:553:
OS_TimerSet() - #3 Interval-too-short (clk_accuracy=10000)

[ PASS] 05.003 ut_ostimer_timerio_test.c:560 - #3 Interval-too-short
[ INFO] ut_ostimer_timerio_test.c:591:
OS_TimerSet() - #1 Nominal condition (clk_accuracy=10000)

[ FAIL] 05.004 ut_ostimer_timerio_test.c:600 - #4 Nominal
[  END] 05 OS_TimerSet          TOTAL::4     PASS::2     FAIL::1 

**System observed on:**
Ubuntu 20.04

**Reporter Info**
ALex Campbell GSFC

jphickey, 2021-04-05:
Likely that whatever this is will be the same as #942 (just hypothesizing) ... but please advise as much as you can about your specific VM config and I can attempt to reproduce on my side.

jphickey, 2021-04-05:
FWIW, today I did open up my VirtualBox install (currently reporting itself as version 6.1.18) and made a new VM with a fresh Ubuntu 20.04 install.  I selected "minimal" install, then did the usual apt-get updates and installed build prerequisites, then pulled down the latest bundle from github and built it.

In the end - I ran `osal_timer_UT` about a dozen times in a row, no failures at all.

So - I'm suspecting this has something to do with configuration.  In particular I'd like to compare the settings for "System" in the VM settings GUI - i.e. amount of RAM and vCPUs assigned to the guest, acceleration options, etc.

Also - One issue I _did_ run into - for some reason, by default, VirtualBox did not immediately enable caching on my virtual disk.  Which meant that the disk was exceptionally slow.  Even just getting the install done was taking way too long - so I forced-off, found the cache setting, enabled it, and restarted the install - which was MUCH faster second time around.

skliper, 2021-08-12:
Note the "1Hz" processing on the system in question is extremely jittery, added a syslog right before the 1Hz elapsed check:
```
    CFE_ES_WriteToSysLog("%s: Good = %lu, Bad = %lu, elapsed sec = %lu.%06lu (%lu)\n", __func__,
                         (unsigned long)CFE_TIME_Global.ToneIntCounter, (unsigned long)CFE_TIME_Global.ToneIntErrorCounter,
                         (unsigned long)Elapsed.Seconds, (unsigned long)CFE_TIME_Sub2MicroSecs(Elapsed.Subseconds),
                         (unsigned long)Elapsed.Subseconds);

    /*
    ** Verify that tone occurred ~1 second after previous tone...
    */
    if (((Elapsed.Seconds == 1) && (Elapsed.Subseconds < CFE_TIME_Global.ToneOverLimit)) ||
        ((Elapsed.Seconds == 0) && (Elapsed.Subseconds > CFE_TIME_Global.ToneUnderLimit)))
```
and a sample result shows huge errors (> 500ms in one case)!  Note on my docker errors are less than 2ms, the default limit is 20ms.  I'm somewhat surprised there isn't at minimum a debug event generated for such poor performance, just a counter in tlm.

```
...
OS_BSP_Initialize():Maximum user msg queue depth = 10
CFE_PSP: Default Reset SubType = 1
CFE_PSP: Default CPU ID = 1
CFE_PSP: Default Spacecraft ID = 66
CFE_PSP: Default CPU Name: cpu1
OS_Posix_GetSchedulerParams():189:Policy 1: available, min-max: 1-99
OS_Posix_GetSchedulerParams():189:Policy 2: available, min-max: 1-99
OS_Posix_TaskAPI_Impl_Init():375:Selected policy 2 for RT tasks, root task = 99
OS_Posix_TaskAPI_Impl_Init():392:Could not setschedparam in main thread: Operation not permitted (1)
CFE_PSP: Instantiated software timebase 'cFS-Master' running at 10000 usec
CFE_PSP: Using POSIX monotonic clock as CFE timebase
CFE_PSP: Using MMAP simulated EEPROM implementation
CFE_PSP: Physical RAM access not implemented
CFE_PSP: I/O Port access not implemented
CFE_PSP: EEPROM Range (2) created: Start Address = 7F11AFECE000, Size = 00080000 Status = 0
CFE_PSP: Starting the cFE with a POWER ON reset.
CFE_PSP: Clearing out CFE CDS Shared memory segment.
CFE_PSP: Clearing out CFE Reset Shared memory segment.
CFE_PSP: Clearing out CFE User Reserved Shared memory segment.
1980-010-08:22:39.94410 CFE_ES_SetupResetVariables: POWER ON RESET due to Power Cycle (Power Cycle).
1980-010-08:22:39.94420 CFE_ES_Main: CFE_ES_Main in EARLY_INIT state
CFE_PSP: CFE_PSP_AttachExceptions Called
1980-010-08:22:39.94580 CFE_ES_Main: CFE_ES_Main entering CORE_STARTUP state
1980-010-08:22:39.94581 CFE_ES_CreateObjects: Starting Object Creation calls.
1980-010-08:22:39.94582 CFE_ES_CreateObjects: Calling CFE_ES_CDSEarlyInit
1980-010-08:22:39.94610 CFE_ES_CreateObjects: Calling CFE_EVS_EarlyInit
1980-010-08:22:39.94613 CFE_EVS_EarlyInit: Event Log cleared following power-on reset
1980-010-08:22:39.94614 CFE_ES_CreateObjects: Calling CFE_SB_EarlyInit
1980-010-08:22:39.94670 CFE_ES_CreateObjects: Calling CFE_TIME_EarlyInit
1980-012-14:03:20.00000 CFE_ES_CreateObjects: Calling CFE_TBL_EarlyInit
1980-012-14:03:20.00073 CFE_ES_CreateObjects: Calling CFE_FS_EarlyInit
EVS Port1 66/1/CFE_EVS 1: cFE EVS Initialized:  cFE DEVELOPMENT BUILD v6.8.0-rc1+dev844 (Codename: Bootes), Last Official Release: cfe v6.7.0
1980-012-14:03:20.15533 CFE_SB_AppInit: Registered 4 events for filtering
EVS Port1 66/1/CFE_SB 1: cFE SB Initialized:  cFE DEVELOPMENT BUILD v6.8.0-rc1+dev844 (Codename: Bootes), Last Official Release: cfe v6.7.0
EVS Port1 66/1/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_SB
EVS Port1 66/1/CFE_ES 1: cFE ES Initialized:  cFE DEVELOPMENT BUILD v6.8.0-rc1+dev844 (Codename: Bootes), Last Official Release: cfe v6.7.0
EVS Port1 66/1/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_ES
EVS Port1 66/1/CFE_ES 2: cFS Versions: cfe v6.8.0-rc1+dev844, osal v5.1.0-rc1+dev586, psp v1.5.0-rc1+dev118. cFE chksm 26583
EVS Port1 66/1/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_ES
EVS Port1 66/1/CFE_ES 91: Version Info: Mission SampleMission, version [unknown]
EVS Port1 66/1/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_ES
EVS Port1 66/1/CFE_ES 91: Version Info: Core Module core_api, version git:v7.0.0-rc1-356-g2fb070a-dirty
EVS Port1 66/1/CFE_ES 91: Version Info: Core Module es, version git:v7.0.0-rc1-356-g2fb070a-dirty
EVS Port1 66/1/CFE_ES 91: Version Info: Core Module evs, version git:v7.0.0-rc1-356-g2fb070a-dirty
EVS Port1 66/1/CFE_ES 91: Version Info: Core Module fs, version git:v7.0.0-rc1-356-g2fb070a-dirty
EVS Port1 66/1/CFE_ES 91: Version Info: Core Module sb, version git:v7.0.0-rc1-356-g2fb070a-dirty
EVS Port1 66/1/CFE_ES 91: Version Info: Core Module tbl, version git:v7.0.0-rc1-356-g2fb070a-dirty
EVS Port1 66/1/CFE_ES 91: Version Info: Core Module time, version git:v7.0.0-rc1-356-g2fb070a-dirty
EVS Port1 66/1/CFE_ES 91: Version Info: Core Module osal, version git:v6.0.0-rc1-322-g7f4ba96
EVS Port1 66/1/CFE_ES 91: Version Info: Core Module psp, version git:v1.5.0-rc1-118-g6d40816
EVS Port1 66/1/CFE_ES 91: Version Info: Core Module msg, version git:v7.0.0-rc1-356-g2fb070a-dirty
EVS Port1 66/1/CFE_ES 91: Version Info: Core Module sbr, version git:v7.0.0-rc1-356-g2fb070a-dirty
EVS Port1 66/1/CFE_ES 91: Version Info: Core Module resourceid, version git:v7.0.0-rc1-356-g2fb070a-dirty
[... 53 lines cut ...]
EVS Port1 66/1/CFE_TIME 21: Stop FLYWHEEL
1980-012-14:03:42.27108 CFE_TIME_Tone1HzISR: Good = 8, Bad = 13, elapsed sec = 1.271086 (1164306363)
1980-012-14:03:43.44344 CFE_TIME_Tone1HzISR: Good = 8, Bad = 14, elapsed sec = 1.172359 (740277986)
[... cut ...]
1980-012-14:03:54.35627 CFE_TIME_Tone1HzISR: Good = 13, Bad = 20, elapsed sec = 1.266624 (1145141790)
EVS Port1 66/1/CFE_TIME 20: Start FLYWHEEL
1980-012-14:03:55.39240 CFE_TIME_Tone1HzISR: Good = 13, Bad = 21, elapsed sec = 1.036125 (155156552)
1980-012-14:03:56.42037 CFE_TIME_Tone1HzISR: Good = 13, Bad = 22, elapsed sec = 1.027975 (120153858)
^C1980-012-14:03:56.87947 CFE_ES_RunExceptionScan: ExceptionID 0x1110000 in TaskID 0: Caught SIGINT
1980-012-14:03:56.87949 CFE_ES_RunExceptionScan: Processor Reset count not reached (0/2)
CFE_PSP: Exiting cFE with PROCESSOR Reset status.
```