nasa/osal #957: Impossible to test OS_CONSOLE_ASYNC false case

skliper, 2021-04-11:
**Is your feature request related to a problem? Please describe.**
Now that OS_CONSOLE_ASYNC is defined locally, it's impossible to functionally test false case or get full branch coverage on `OS_ConsoleCreate_Impl`:
https://github.com/nasa/osal/blob/ba0ac40d3b58b4d12f32a2222c8c8ed2fb663c65/src/os/vxworks/src/os-impl-console.c#L41-L48

https://github.com/nasa/osal/blob/ba0ac40d3b58b4d12f32a2222c8c8ed2fb663c65/src/os/vxworks/src/os-impl-console.c#L146-L148

**Describe the solution you'd like**
Either need to be able to exercise this option or remove it.

**Describe alternatives you've considered**
None

**Additional context**
Prevents full branch coverage

**Requester Info**
Jacob Hageman - NASA/GSFC

jphickey, 2021-04-12:
I'm trying to recall why this was removed from osconfig.h ... It is still relevant/preferable to run in this mode for debugging situations.  Many times OS_printf is used for debugging and if the actual output is decoupled, it can change the order with which things appear.  It also saves the memory cost of another task+stack in constrained environments.

If I understand correctly the issue here is limited to the "create_impl" function only - since it can't easily be tested with the flag as false.  Maybe a compromise solution would be to move the boolean from the impl layer to the shared layer, which would allow the impl coverage test to set it both ways at least.

skliper, 2021-04-12:
I'd rather make it an option that can be tested functionally or remove it.  Coverage testing with overrides doesn't prove all the logic works together.

Option - ifndef make it async=true.

skliper, 2021-04-12:
Or relate it to OS_DEBUG or similar, and/or note the expected context/reasoning.

jphickey, 2021-04-12:
I consider it similar to the shell switch or something like that - for FSW the user chooses which way to configure it and then it gets compiled in that config, but for coverage test - due to the design and its "backdoor" methods of running things directly, it can still test it both ways, regardless of how its configured in FSW.

Right now even that's not possible because of the proximity to the definition and the runtime check.  If they were in separate units, this would be easily doable.

Can also resurrect this as an osconfig option.  As I said, I don't recall why it was removed, but might have been before osconfig.h was converted to osconfig.cmake.  In the current design there shouldn't be a problem of having this option, and defaulting it to true.

I'd rather not tie it to OS_DEBUG - although that's one justification for changing it - simply saving the cost of an extra task+stack might be worthwhile to some users.

PR #959 (2021-04-28): Fix #957, move async console option
**Describe the contribution**
Puts the "async" option into the shared layer instead of the impl layer.

This allows both options to be coverage tested and also allows a bit more of the logic to be common instead of duplicated in the 3 implementations.

This also adds back an osconfig option to allow the user to elect this mode at configuration time.

Fixes #957 

**Testing performed**
Build and sanity check CFE with `OSAL_CONFIG_CONSOLE_ASYNC` set each way
Confirm coverage test also exercises both runtime paths, regardless of compile time config.
Confirm (vi [... cut]
commit c518e3f3 (2021-04-28): Merge pull request #959 from om jphickey/fix-957-consoletask-async-option

Fix #957, move async console option
commit ccf9215f (2021-04-28): Fix #957, move async console option

Puts the "async" option into the shared layer instead of the impl layer.
This allows both options to be coverage tested and also allows a bit more
of the logic to be common instead of duplicated in the 3 implementations.

This also adds back an osconfig option to allow the user to elect this
mode at configuration time.
commit 811d48ac (2021-04-28): Merge pull request #959 from om jphickey/fix-957-consoletask-async-option

Fix #957, move async console option
closed 2021-04-29 by astrogeco