nasa/osal #970: return for a timeout is negative 4

excaliburtb, 2021-04-26:
https://github.com/nasa/osal/blob/afb5f7ba3ea48823e44490ce4258aa92a073087f/src/os/shared/src/osapi-file.c#L208

Not sure which layer of abstraction to report this but the comment states that a timeout should return 0.  However, on Linux/portable BSD... the eventual call to OS_DoSelect is returning OS_ERROR_TIMEOUT (-4) and flowing that all the to the OS_TimedRead API. I would prefer it to be "caught" and returned as 0 at some point but.. more than anything.. just need the comment to match.

jphickey, 2021-04-26:
Is this the comment you are referring to? https://github.com/nasa/osal/blob/afb5f7ba3ea48823e44490ce4258aa92a073087f/src/os/inc/osapi-file.h#L224

Note that a read of size 0 typically indicates an EOF condition.  POSIX says this about read(): 

    No  data transfer shall occur past the current end-of-file. If the starting position is at or after the end-of-file, 0 shall be returned.

This is a paradigm that OSAL is borrowing/passing through here, so returning `OS_ERROR_TIMEOUT` on timeout is preferable in order to distinguish this as a true timeout, rather than EOF.  This is particularly important for sockets where the remote end has closed the connection.

I will update the documentation to correct this.   We should probably also add a note that a read of size 0 is the EOF indicator.

excaliburtb, 2021-04-26:
yep, the api comments need to not say timeout == 0.

skliper, 2021-04-27:
Reference to erroneous line:
https://github.com/nasa/osal/blob/afb5f7ba3ea48823e44490ce4258aa92a073087f/src/os/inc/osapi-file.h#L217

excaliburtb, 2021-04-27:
and just to double-check, double-confirm... for TCP sockets the "replacement" for send and recv calls is OS_TimedRead and OS_TimeWrite, correct?

PR #972 (2021-04-28): Fix #970, update documentation for read/write
**Describe the contribution**

Update the API documentation for OS_read/write/TimedRead/TimedWrite. Clarify that zero will be returned on EOF condition, and in the case of the timed API calls, the OS_ERR_TIMEOUT status code will be returned if the timeout expired without the handle becoming readable/writable during that time.

This is intended behavior such that it allows the caller to differentiate between a handle which is in a state where it cannot read/write any more data (e.g. at EOF, or a pipe/socket with remote end closed) and a handle which [... cut]
commit dcb9f832 (2021-04-28): Fix #970, update documentation for read/write

Update the API documentation for OS_read/write/TimedRead/TimedWrite.
Clarify that zero will be returned on EOF condition, and in the
case of the timed API calls, the OS_ERR_TIMEOUT status code will
be returned if the timeout expired without the handle becoming
readable/writable during that time.

This is intended to allow the caller to differentiate between a
handle which is in a state where it cannot read/write data (e.g.
at EOF, or a pipe/socket with remote end closed) and handle which
is simply idle or busy.
commit f9e879e0 (2021-04-28): Merge pull request #972 from jphickey/fix-970-readwrite-docs

Fix #970, update documentation for read/write
PR #975 (2021-04-29): osal Integration candidate: 2021-04-27
## Description

### PR #972

Fix #970, update documentation for read/write

[docs] Clarifies that that zero will be returned on EOF condition in the API documentation for OS_read/write/TimedRead/TimedWrite. In the case of the timed API calls, the `OS_ERR_TIMEOUT` status code will be returned if the timeout expired without the handle becoming readable/writable during that time.

### PR #966

Fix #832, add "handler" feature to utassert stub API

Addresses a shortcomings in the UT Assert hook functions. Namely the assumed return type of int32 which is not al [... cut]
closed 2021-04-29 by astrogeco