nasa/osal #1007: Remove test run from CodeQL workflow

astrogeco, 2021-05-12:
**Describe the bug**
CodeQL workflow is running unit tests which is unnecessary. 

**To Reproduce**
See https://github.com/nasa/osal/actions/runs/835716057

**Expected behavior**
Code QL workflow should only run enough of the build steps to run the CodeQL analysis

**Code snips**
https://github.com/nasa/osal/blob/9756b0396807a7dcd64230f01f38e4cad5f3d4f7/.github/workflows/codeql-build.yml#L49-L50

**System observed on:**
github actions 

**Additional context**
Ensure that new CodeQL workflow in #986 doesn't have the "run tests" step

skliper, 2021-05-12:
If you remove it from here it needs to be added in somewhere else or as a new workflow.  It was easy here since everything was built already.  Could be done parallel.

astrogeco, 2021-05-12:
Aren't they running in https://github.com/nasa/osal/runs/2569594615?check_suite_focus=true ?

skliper, 2021-05-12:
Your reference is just the local unit tests for OSAL, the run referenced in this issue is the entire suite of cFS unit tests (with the cFS configuration).

skliper, 2021-05-12:
Note the full run will likely fail when there are dependencies, which I think is a GOOD thing (catches when we've missed something, notifies as part of the merge process that extra care is needed).  As long as the PR justifies why it fails that's fine by me.

astrogeco, 2021-05-12:
ah! the output looked similar so I thought they were the same. Still think they should live in separate workflows but I do get the point of using the existing build to avoid duplicating the build step. I edited the original text and added an alternative resolution