nasa/osal #1086: network-api-test timeout still too short for MCP750

jphickey, 2021-06-25:
**Describe the bug**
The console/printf operations on MCP750 take a considerable amount of time (only 9600 baud serial).  It appears some extra assert statements added in this test have made it so even 1000ms is not sufficient.  (looks like the ServerFn might be timing out on a write while the client is printing a bunch of asserts, which then causes an error later when the client finally gets to the read).

**To Reproduce**
Run test on MCP750, observe unexpected timeout error:

    [ FAIL] 04.040 network-api-test.c:680 - OS_TimedRead() (-4) == 4

**Expected behavior**
Test should pass

**System observed on:**
MCP750 VxWorks 6.9

**Reporter Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2021-06-25:
Bumping the timeout up to 4000ms seems to fix the issue.... so this is what I'd propose.

skliper, 2021-06-25:
Concur.

PR #1092 (2021-07-02): Fix #1086, increase timeout in network-api-test
**Describe the contribution**
The number of asserts being printed was still taking longer than 1000ms on a 9600 baud serial link, so increasing the timeout allows tests to pass.

Fixes #1086

**Testing performed**
Run network-api-test

**Expected behavior changes**
Test now passes

**System(s) tested on**
MCP750 / VxWorks 6.9

**Contributor Info - All information REQUIRED for consideration of pull request**
Joseph Hickey, Vantage Systems, Inc.
commit 55cb7b28 (2021-07-02): Fix #1086, increase timeout in network-api-test

The number of asserts being printed was still taking longer than 1000ms
on a 9600 baud serial link, so increasing the timeout allows tests to
pass.
commit 931c4698 (2021-07-02): Merge pull request #1092 from jphickey/fix-1086-network-api-timeout

Fix #1086, increase timeout in network-api-test
PR #1098 (2021-07-07): osal Integration candidate: 2021-06-29
## Context 

Part of <https://github.com/nasa/cfs/pull/287>

## Tests

<https://github.com/nasa/cfs/pull/287/checks>
<https://github.com/nasa/osal/pull/1098/checks>

## Authors

@jphickey 
@skliper 

## Description

### PR #1081
Fix #1080, add range to OS_TaskDelay checks

### PR #1085
Fix #1084, Return `OS_OBJECT_ID_UNDEFINED` for root task ID on RTEMS

### PR #1090
Fix #1082, increase UT symbol dump size limit

### PR #1091
Fix #1087, do not register RTOS timer for external sync

### PR #1093
Fix #1088, avoid task delete during UtPrintf 

### PR #1097
F [... cut]
closed 2021-07-07 by astrogeco