nasa/osal #1094: VxWorks osal_file_UT test does not pass with default config

jphickey, 2021-06-25:
**Describe the bug**
The osal_file_UT uses a small ramdisk as its temporary file system to run the tests.  However, with the default configuration of `OSAL_CONFIG_MAX_NUM_OPEN_FILES` at 50, one gets the following error:

    OS_FileOpen_Impl():114:open(RAM3:0/tmpFile26.txt): errno = 0x380004

This errno correlates to `S_dosFsLib_NO_FREE_FILE_DESCRIPTORS` 

Reducing the value of `OSAL_CONFIG_MAX_NUM_OPEN_FILES` to 24, the test passes OK

**To Reproduce**
Run osal_file_UT on MCP750 / VxWorks 6.9 with default config

**Expected behavior**
Test should pass with default config

**Code snips**
If applicable, add references to the software.

**System observed on:**
MCP750 / VxWorks 6.9

**Additional context**
The RAMDISK only has 64 blocks, so this might be related.  The other option, if acceptable, might be to document in the release notes that this particular test on this platform requires the OSAL_CONFIG_MAX_NUM_OPEN_FILES to be reduced from its default value.

**Reporter Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2021-06-25:
NOTE - I did do a quick test on the theory of increasing the size of the ramdisk, but it doesn't help.  I increased the size of the ramdisk from 64 blocks to 132 blocks, but it still failed at the same file (RAM3:0/tmpFile26.txt).  So there appears to be some sort of limitation on VxWorks RAM disks around 25 concurrent open files.

skliper, 2021-06-28:
Adding milestone to make sure we document the restriction related to testing on the MCP750/VxWorks target.