nasa/osal #1095: osal_timer_UT inconsistent on MCP750 / VxWorks

jphickey, 2021-06-25:
**Describe the bug**
The timer test does not have consistent results when run on the MCP750 + VxWorks 6.9 platform

**To Reproduce**
Run "osal_timer_UT" and occasionally a failure occurs in the "nominal" timer case:

```
[ PASS] 05.006 ut_ostimer_timerio_test.c:468 - OS_TimerSet(g_timerIds[4], startTime, intervalTime) (0) == OS_SUCCESS (0)
[ FAIL] 05.007 ut_ostimer_timerio_test.c:475 - 4# Nominal - callback status -1
```

**Expected behavior**
Test should pass reliably

**Code snips**
This is partly related to the fact that the MCP750 implementation has something wrong with its timer implementation.  The test reports these warnings:

```
OS_TimeBaseSet_Impl():576:WARNING: timer 524301 start_time requested=1000us, configured=16666us
OS_TimeBaseSet_Impl():582:WARNING: timer 524301 interval_time requested=500000us, configured=516666us
```

This means there is increased jitter in the callbacks, which is then compounded by the rather crude delta check:

https://github.com/nasa/osal/blob/64a6b31b5448a6301c64b001e043d4c7d9570131/src/unit-tests/ostimer-test/ut_ostimer_test.c#L104

**System observed on:**
MCP750 / VxWorks 6.9

**Additional context**
Other timer tests which average the counts over total time pass fine.  But this one checks the delta on every interval, so one case of too much jitter and it fails.

**Reporter Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2021-06-25:
I believe the issue here is specific to the MCP750, not really a bug in the test itself.  Because it is configuring the kernel timer for 516666us instead of 500000us, the test case sees an absurd amount of jitter in the resulting callbacks.  If the kernel could configure a 500000us timer properly, this test would run fine most likely.

skliper, 2021-06-28:
Added milestone to ensure it gets documented in release notes (no "fix" required, it's related to resolution of the kernel timer on the MCP750).