nasa/osal #1172: Check Return Value for setsockopt

ArielSAdamsNASA, 2021-09-29:
**Describe the bug**
Calling setsockopt(impl->fd, 1, 2, &os_flags, 4U) without checking return value. This library function may fail and return an error code.

**Expected behavior**
Check return value for setsockopt

**Code snips**
https://github.com/nasa/osal/blob/42af0f73c952b4e4c7fbc06c97d354a1430f4ac3/src/os/portable/os-impl-bsd-sockets.c#L199

**System observed on:**
Coverity: https://scan.coverity.com/projects/arielsadamsnasa-cfs-jsf-rules?tab=overview

**Additional context**
Possible solution 
```
if (setsockopt(impl->fd, SOL_SOCKET, SO_REUSEADDR, &os_flags, sizeof(os_flags)) == 0)
{
    setsockopt(impl->fd, SOL_SOCKET, SO_REUSEADDR, &os_flags, sizeof(os_flags));
}
```

**Reporter Info**
Ariel Adams, ASRC Federal

himanshu007-creator, 2021-09-29:
hi i would like to work on this issue

ArielSAdamsNASA, 2021-09-29:
@himanshu007-creator  Thank you for the participation! I assigned you to this issue.

skliper, 2022-10-14:
@ArielSAdamsNASA @himanshu007-creator - I'm not sure this solution solves an actual issue.  This suggestion just does the same call again if the first fails which doesn't seem to be any better (odds are it'll just fail again).  Although it is a static analysis warning, it's already documented (in comments) that if the call fails there isn't a need to take any further action.  I'd recommend just closing this issue and the related pull request.

closed 2025-03-11 by dzbaker

himanshu007-creator, 2025-03-11:
Got it