nasa/osal #1173: Copy of Overlapping Memory in snprintf

ArielSAdamsNASA, 2021-09-29:
**Describe the bug**
In the call to function snprintf, the object pointed to by argument local->volume_name may overlap with the object pointed to by argument local->system_mountpt.

local->system_mountpt is 64 bytes off of the address of OS_filesys_table[OS_ObjectIndexFromToken(token)].

local->volume_name is 32 bytes off of the address of OS_filesys_table[OS_ObjectIndexFromToken(token)]

**Expected behavior**
A clear and concise description of what you expected to happen.

**Code snips**
https://github.com/nasa/osal/blob/42af0f73c952b4e4c7fbc06c97d354a1430f4ac3/src/os/posix/src/os-impl-filesys.c#L175

**System observed on:**
Coverity: https://scan.coverity.com/projects/arielsadamsnasa-cfs-jsf-rules?tab=overview

**Reporter Info**
Ariel Adams, ASRC Federal

Jkillelea, 2023-01-14:
Hi all,

This bug just cropped up for me again, at commit 419c673b9d982bf0bd34546a7568201c6bcf8b62. 

The specific error is:

```
/home/[user]/projects/cFS/osal/src/os/posix/src/os-impl-filesys.c:173:9: error: ‘snprintf’ argument 5 overlaps destination object ‘OS_filesys_table’ [-Werror=restrict]
  173 |         snprintf(local->system_mountpt, sizeof(local->system_mountpt), "%s/osal:%s", tmpdir, local->volume_name);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

sunbrst, 2023-01-22:
Confirmed that I'm seeing this issue too (failing a build of cFS - https://github.com/nasa/cFS). Has anyone found a commit that works for this?

jphickey, 2023-01-23:
I've looked at this before but it doesn't seem like a real issue.  Yes - the `volume_name` and `system_mountpt` are members of the same struct, but they don't overlap.  As long as  `volume_name` is properly null-terminated within `OS_FS_VOL_NAME_LEN` bytes (32 by default, which it should be, I can't see any reason why it would not be) then this will not access overlapping memory.

I think the reason the compiler reports this warning is because if it was not null-terminated, it could theoretically write up to `OS_MAX_LOCAL_PATH_LEN` bytes into the destination, which is 128 by default.  Because the pointers are only 32 bytes apart, this is why it gets flagged.  

I am thinking of a possible workaround but in the meantime I'd suggest just adding `-Wno-restrict` to your compiler options.

closed 2023-01-26 by dzbaker

smparekh, 2023-01-29:
Thanks folks, I can confirm building of cFS on the latest commit.

commit 963d3577 (2023-01-26): Merge pull request #1355 from jphickey/fix-1173-mountpt-string-append

Fix #1173, separate append on volume_name to system_mountpt
commit a67f5e20 (2023-01-26): Fix #1173, separate append on volume_name to system_mountpt

Generating the system_mountpt string via a single call to snprintf
triggered a compiler warning about overlapping memory.  However this
does not seem like a real warning, as the volume_name should always
be null terminated before the overlap would be possible.

By separating this to be a separate append of the volume_name along with
an explicit size check to ensure the buffers indeed do not overlap, this
avoids the warning.
PR #1355 (2023-01-26): Fix #1173, separate append on volume_name to system_mountpt
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/osal/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
Generating the system_mountpt string via a single call to snprintf triggered a compiler warning about overlapping memory. [... cut]