nasa/osal #1194: Give the UtAssert_<type here>_<comparison here> macros versions that can add user 'desc'

asgibson, 2021-11-22:
**Is your feature request related to a problem? Please describe.**
Output is generic and some additional information is desired.  e.g. Turn '*received.RunStatus (3) == *test->expected.RunStatus (3)' into 'CFE_ES_RunLoop:*received.RunStatus (3) == *test->expected.RunStatus (3)'

**Describe the solution you'd like**
Take each of the macros in [utassert.h](https://github.com/nasa/osal/blob/main/ut_assert/inc/utassert.h) that are designed for these checks and add a second version that has _WITH_DESC or something of that nature.

**Describe alternatives you've considered**
Already written my own locally needed versions.

**Additional context**
As an example for UT_UNT32_EQ would look like extended:
```
#define UtAssert_UINT32_EQ_WITH_DESC(actual, ref, desc)                                                                           \
    UtAssert_GenericUnsignedCompare((uint32)(actual), UtAssert_Compare_EQ, (uint32)(ref), UtAssert_Radix_DECIMAL, \
                                    __FILE__, __LINE__, desc, #actual, #ref)
```

**Requester Info**
Alan Gibson
NASA/GSFC 587

asgibson, 2021-11-23:
In truth, were I to have the _WITH_DESC versions, I would not use anything else. Mostly because I believe in unit test verbosity.  I just found it odd that all of these macros send "" (empty string) as the desc.  This led to the thought it would be nice to have ones that could actually use that feature. I can provide my own implementations, but thought they may prove useful to others.

I do not like the UtPrintf option due to additional code line and more test output:
```
UtPrintf("CFE_EVS_SendEvent:");
UtAssert_UINT32_EQ(received.EventID, test->expected.EventID);
```
becomes
```
[ INFO] example.c:54:CFE_EVS_SendEvent:
[ PASS] 01.003 example.c:55 - received.EventID (6) == test->expected.EventID (6)
```
while consolidating it to the one call
```
UtAssert_UINT32_EQ_WITH_DESC(received.EventID, test->expected.EventID, "CFE_EVS_SendEvent: ");
```
becomes
```
[ PASS] 01.003 example.c:54 - CFE_EVS_SendEvent: received.EventID (6) == test->expected.EventID (6)
```

jphickey, 2021-11-23:
Note that in most cases, the logging of event IDs like above would generally come after asserting on the return code from a call to some function (e.g. CFE_EVS_SendEvent), such as:

    UtAssert_INT32_EQ(CFE_EVS_SendEvent(0, CFE_EVS_EventType_INFORMATION, "OK"), CFE_SUCCESS);
    UtAssert_UINT32_EQ(received.EventID, test->expected.EventID);

When reading the logs you'll see two messages, so you mostly get this context "for free" if reading the logs.  I do concur that the downside of including extra free-form tags is that it makes the log output harder to digest by a script.  Being able to regex-match the test log outputs has proven helpful in the past, this but depends on having a consistent formatting in the output.

The extra description tag on the internal function was probably just there originally to match old/existing test outputs.  In some ways, less is more in this regard, because its fixed information, not runtime state information.  Priority should be to capture the runtime data that might be ephemeral (such as if it the event ID was an uninitialized value, or corrupted value, etc) so the developer who is tasked with debugging it can know that information, in case it isn't readily reproducible.  

The context info/function that was previously called is very obvious by opening the source file and going to the line number of the test that failed, so it is not really important to have that info directly in the log file, if its in the source file.

asgibson, 2021-11-23:
I do not agree with the "most cases" assumption as none of my unit tests would ever look like that. My use case as I described is not a test when I am calling CFE_EVS_SendEvent, that function is called by my code under test and I am verifying what it received. I also do not agree with "less is more in this regard" because the goal of unit testing should not be the obfuscation of results, but instead clarity. Readability of results is important to me because I develop with test driven design. I am constantly running the tests as the production code is developed, but I am also concerned with coming back much later and having good outputs describing what is occurring at runtime.

Again, I advocate for verbosity in unit test output. If it is not desired to give users a more verbose option here because other ways exist, then so be it. However, I still think it would be a valuable addition. Plus, maintaining these extra macros should just be a little regex work when things change.