nasa/osal #1199: New compile warnings with GCC 11.x

jphickey, 2022-01-06:
**Describe the bug**
The newer versions of GCC report new warnings.  These appear to be mostly false alarms, but easy to fix nonetheless.

**To Reproduce**
Build OSAL using recent GCC

**Expected behavior**
Should compile without warnings or errors

**Code snips**

```
osal/src/unit-test-coverage/vxworks/src/coveragetest-tasks.c: In function ‘Test_OS_TaskDetach_Impl’:
src/unit-test-coverage/vxworks/src/coveragetest-tasks.c:151:28: error: ‘token’ may be used uninitialized [-Werror=maybe-uninitialized]
src/unit-test-coverage/shared/src/coveragetest-clock.c: In function ‘Test_OS_SetLocalTime’:
src/unit-test-coverage/shared/src/coveragetest-clock.c:55:26: error: ‘time_struct’ may be used uninitialized [-Werror=maybe-uninitialized]
osal/src/unit-tests/oscore-test/ut_oscore_task_test.c: In function ‘UT_os_task_getid_by_sysdata_test’:
osal/src/unit-tests/inc/ut_os_support.h:112:28: error: ‘sysdata’ may be used uninitialized [-Werror=maybe-uninitialized]
```

**System observed on:**
Ubuntu 21.10, gcc 11.2.0

**Reporter Info**
Joseph Hickey, Vantage Systems, Inc.

kurtjd, 2022-01-08:
I wouldn't mind tackling this, however I am having trouble replicating the issue. I'm using the same version of GCC on Pop!_OS, but building OSAL standalone doesn't seem to output any errors. My method of building is the one listed in the OSAL Configuration Guide:

```
$ mkdir build
$ cd build
$ cmake \
    -DCMAKE_BUILD_TYPE=debug \
    -DENABLE_UNIT_TESTS=true \
    -DOSAL_SYSTEM_BSPTYPE=generic-linux \
    -DOSAL_CONFIG_DEBUG_PERMISSIVE_MODE=true \
     /path/to/osal/source
```
I also tried adding `-DOSAL_USER_C_FLAGS=-Werror` but this didn't seem to provide any error output either.

skliper, 2022-01-10:
My guess is these warnings were observed when compiling OSAL from within the bundle context (https://github.com/nasa/cFS).  There's additional warnings enabled here (if/when you copy in the sample_defs to the bundle context):
https://github.com/nasa/cFE/blob/c161cf04fe7c5fd2f2a533ac1d7522b31191eab6/cmake/sample_defs/arch_build_custom.cmake#L28-L37

jphickey, 2022-01-10:
It requires a very recent version of GCC (11+).  IIRC Pop OS is based on 20.04 so that's too old.

I actually have fix already, I patched it up so I could build on this system (Ubuntu 21.10), but haven't made a PR yet.

kurtjd, 2022-01-10:
Oh ok, yeah I have the latest version of GCC but I wasn't building as part of the bundle. Thanks for the clarification!

commit 17b71636 (2022-01-19): Merge pull request #1203 from jphickey/fix-1199-gcc11-warnings

Fix #1199, correct warnings on gcc11

 # Please enter a commit message to explain why this merge is necessary,
commit e51cbe33 (2022-01-19): Fix #1199, correct warnings on gcc11

These new warnings are detected by the new compiler. They are all places
in unit test where an uninitialized value is passed via "const" pointer
into a unit under test.  By definition a "const" pointer is always an
input.

While the warning is pedantically true - should not pass an uninitialized
struct as an input - these were all unit tests where the object value
is a don't-care value, so it does not matter.  But to fix the warning,
simply need to initialize a value before making the call.
PR #1203 (2022-01-21): Fix #1199, correct warnings on gcc11
**Describe the contribution**
- Fix #1199
These new warnings are detected by the new compiler. They are all places in unit test where an uninitialized value is passed via "const" pointer into a unit under test.  By definition a "const" pointer is always an input.

While the warning is pedantically true - should not pass an uninitialized struct as an input - these were all unit tests where the object value is a don't-care value, so it does not matter.  But to fix the warning, simply need to initialize a value before making the call.

Fixes #1199

**Testing p [... cut]
closed 2022-01-21 by astrogeco