nasa/osal #1205: Optional support for symbol table dump and module address/size info on POSIX systems (adds GNU extension dependence)

skliper, 2022-01-11:
**Is your feature request related to a problem? Please describe.**
Currently checksum app does not work on systems that use os-impl-posix-dl-symtab.c, nor is OS_SymbolTableDump supported.

**Describe the solution you'd like**
Would add GNU extension dependence, but dl_iterate_phdr() could be used to get the information required for both symbol table dump and module address/size.  Example for getting module address here: https://stackoverflow.com/questions/19451791/get-loaded-address-of-a-elf-binary-dlopen-is-not-working-as-expected

**Describe alternatives you've considered**
None

**Additional context**
We've avoided GNU extension dependence, but if supported on the desired system it would be helpful to have the option for this functionality to work.  Another use case is when simulating a system that does have these functions implemented, it's likely desirable to support the capability.

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2022-01-12:
Updated suggestion - Add OS_NotifyEvent OS_EVENT_READ_MODULE_INFO and OS_EVENT_DUMP_SYMBOL_TABLE or similar and put the platform specific logic in the PSP.

skliper, 2022-01-25:
Added @acudmore to help with RTEMS approach/concept.  Although toe OSAL impact is probably the same (add the OS_NotifyEvent calls and fix in the PSP).

astrogeco, 2022-03-01:
@dmknutsen suggested using these two tools 

https://www.mankier.com/1p/nm
https://docs.rtems.org/releases/4.5.1-pre3/toolsdoc/binutils-2.10-docs/binutils/binutils00002.html

acudmore, 2022-03-01:
I started reviewing the existing code and OS_NotifyEvent framework.

For RTEMS, with Eric's pull request (#1221 ) we have a complete os-impl-loader.c
To complete RTEMS support, we need to create a non-standard os-impl-symtab.c, since I believe the code will be RTEMS specific and will not work with POSIX calls. I don't think it will make sense to use OS_NotifyEvent calls to the PSP, since this code should work on all RTEMS targets, and would have to be duplicated in all RTEMS based PSPs. 
There is an RTEMS shell command that dumps all symbols to the console, so I can use it as a base for the implementation.

I still have to look at POSIX.

acudmore, 2022-03-01:
> @dmknutsen suggested using these two tools
> 
> https://www.mankier.com/1p/nm https://docs.rtems.org/releases/4.5.1-pre3/toolsdoc/binutils-2.10-docs/binutils/binutils00002.html

Definitely useful on the host, I use them all of the time to dump the symbols from executable images, but I don't think these will work on the targets. Nice tools to be familiar with though!